home-assistant/operating-system #225: Unable to access host through SSH

robbinjanssen, 2018-10-22:
I am trying to access the host system via SSH (port 2222). I've followed all instructions on the [Debugging Hass.io](https://developers.home-assistant.io/docs/en/hassio_debugging.html) page but have not been able to login. I'm using the same key as I'm using for the port 22 connection with the SSH addon, so the key is not the problem. Any tips? Is there any specific log message I need to see when I import from USB?  

Note; my USB stick is named "CONFIG", it is formatted as FAT and only has the `authorized_keys` file on it. Key and `authorized_keys` file have been created on a macbook using the terminal.

**Version:** HassOS 1.11

robbinjanssen, 2018-10-22:
Okay, i'm stupid, trying to connect to port 2222 instead of 22222.

Shame on me!

closed 2018-10-22 by robbinjanssen

mhaas, 2022-08-13:
@robbinjanssen Four years later, I find this issue. Exact same problem here - PEBKAC. Thanks for sharing!