home-assistant/operating-system #226: Can't install homeassistant/qemux86-64-homeassistant:0.80.3

naamah75, 2018-10-25:
I'm trying to update the HASSos from 0.79.3 to 0.80.3 but I got the following error (from logs):

```
18-10-25 09:02:04 INFO (SyncWorker_6) [hassio.docker.interface] Update Docker 0.79.3 with homeassistant/qemux86-64-homeassistant:0.80.3
18-10-25 09:02:04 INFO (SyncWorker_6) [hassio.docker.interface] Pull image homeassistant/qemux86-64-homeassistant tag 0.80.3.
18-10-25 09:05:34 ERROR (SyncWorker_6) [hassio.docker.interface] Can't install homeassistant/qemux86-64-homeassistant:0.80.3 -> 404 Client Error: Not Found ("no such image: homeassistant/qemux86-64-homeassistant:0.80.3: No such image: homeassistant/qemux86-64-homeassistant:0.80.3").
18-10-25 09:05:34 WARNING (MainThread) [hassio.homeassistant] Home Assistant is already running!
18-10-25 09:05:34 INFO (MainThread) [hassio.homeassistant] Successful run Home Assistant 0.80.3
```
Despite the last line of the log, really I continue to run the 0.79.3 version even If I restart the supervisor or the entire system.

geelenbert, 2018-10-29:
same issue here:

`18-10-29 20:38:52 INFO (MainThread) [hassio.homeassistant] Update Home Assistant to version 0.81.1
18-10-29 20:38:52 INFO (SyncWorker_2) [hassio.docker.interface] Update Docker 0.79.2 with homeassistant/qemux86-64-homeassistant:0.81.1
18-10-29 20:38:52 INFO (SyncWorker_2) [hassio.docker.interface] Pull image homeassistant/qemux86-64-homeassistant tag 0.81.1.
18-10-29 20:39:46 ERROR (SyncWorker_2) [hassio.docker.interface] Can't install homeassistant/qemux86-64-homeassistant:0.81.1 -> 404 Client Error: Not Found ("no such image: homeassistant/qemux86-64-homeassistant:0.81.1: No such image: homeassistant/qemux86-64-homeassistant:0.81.1").
18-10-29 20:39:46 WARNING (MainThread) [hassio.homeassistant] Home Assistant is already running!
18-10-29 20:39:46 INFO (MainThread) [hassio.homeassistant] Successful run Home Assistant 0.81.1`

can't update to newer home assistant version.

naamah75, 2018-10-30:
Same issue when trying to update to 0.81.1

geelenbert, 2018-10-31:
Same issue when trying to update to 0.8.1.2

`18-10-31 11:22:28 INFO (SyncWorker_7) [hassio.docker.interface] Update Docker 0.79.2 with homeassistant/qemux86-64-homeassistant:0.81.2
18-10-31 11:22:28 INFO (SyncWorker_7) [hassio.docker.interface] Pull image homeassistant/qemux86-64-homeassistant tag 0.81.2.
18-10-31 11:23:21 ERROR (SyncWorker_7) [hassio.docker.interface] Can't install homeassistant/qemux86-64-homeassistant:0.81.2 -> 404 Client Error: Not Found ("no such image: homeassistant/qemux86-64-homeassistant:0.81.2: No such image: homeassistant/qemux86-64-homeassistant:0.81.2").
18-10-31 11:23:21 WARNING (MainThread) [hassio.homeassistant] Home Assistant is already running!
18-10-31 11:23:21 INFO (MainThread) [hassio.homeassistant] Successful run Home Assistant 0.81.2`

naamah75, 2018-10-31:
Yes, same problem with 0.81.2  ...

naamah75, 2018-11-25:
Solved for me with 1.13

closed 2018-11-25 by naamah75

rlust, 2018-12-01:
Same issue trying to update to 0.83 0.83.1 0.83.2