home-assistant/operating-system #227: Not autobooting. Stops at boot menu

bhaap, 2018-10-25:
Hi,

Im running hassos in virtualbox and had forgotten to increase the partition size before running the VM.

When the computer eventually crashed because of no hd space I increased the partition size. All was well until i realized that the VM would not autoboot anymore. 
I always end up at the Boot menu.
I can continue by pressing  "Boot System 0". 

But Autoboot does not work.

Any way to repair the autoboot?

[image]

WhiteNoseThunderPants, 2018-10-25:
@bhaap Sorry to jump on your issue, but I also did not change the disk size of the VM before using it. I have run out of space and it will no longer boot. I changed the disk size in Proxmox, but how do I get into the system to change the partition size?

WhiteNoseThunderPants, 2018-10-25:
Ok, so I used gparted to extend the partition and now I have the same problem as @bhaap on booting.

bhaap, 2018-10-26:
That is exactly how i increased the partition size.
wonder if booting on a disc drive destroyed the autoboot.

bhaap, 2018-11-01:
Anyone have any insight in how to fix this?

pvizeli, 2018-11-01:
what is the output of `rauc info` on cmd after booting up?

bhaap, 2018-11-06:
[image]

closed 2019-04-23 by stale[bot]

marine1988, 2020-03-30:
i have the same problem!

vin901, 2022-04-30:
I managed to fix this issue, but it may not apply to everyone.

I was running HA on VMware with Dell PowerEdge 1950 server, and sometimes I would get the VMware error popup "vim.fault.GenericVmConfigFault" when I shutdown the HA host. HA would also randomly get corrupted filesystem errors on the console output after it's been running for days. But one day it would keep showing the boot menu like the one posted by OP.

If I selected "Boot System 0" it boots HASSIO 7.5 but then reboots into the boot menu again.
If I selected "Boot System 1" it boots HASSIO 7.6 but then also reboots into the boot menu again.

SOLUTION:

So I was upgrading memory on the server and I noticed that there was a **leaking capacitor** on the **SAS raid controller** board that connects to the 2 SATA drives at the front of the chassis. 

I replaced both capacitors (1500uF 6.3V) on the board just to be sure, and booted up the server.
The boot menu still showed up, but after selecting "Boot System 1", it booted and HA started up fine! :-)

It seems the leaked capacitor was causing data corruption when it was reading the drives or something. So even if you don't have the same hardware, maybe your harddrive is corrupted or the controller of your harddrive is going bad.

Anyways on another same spec server, when both capacitors went bad the bios would give you a "PCIe Training Error. Internal Storage Slot error. E2022 Post Fail.", so in my case this particular server running HA was starting to go bad but not completely bad enough that it still manages to pass the BIOS boot checks.

Good luck!

thegreenpan, 2022-05-05:
I had the same problem. I removed the Optical drive from the boot order, and from the storage options, and it is booting up now.

Maybe it might be another issue, but I paused the virtual machine just before it stalled on the autoboot. 
There was some error about it not being able to find a file in the optical drive. 

Since I don't use an optical drive in the machine, I just removed it.

boots up fine now.