home-assistant/operating-system #233: HASSOS 1.12 VMDK

cooperj, 2018-10-29:
Tried running in VirtualBox, reported no bootable device. Virtual Disk Corrupt?

pvizeli, 2018-10-29:
Hmm should not. I rebuild it

MwC-Trexx, 2018-10-29:
The 1.12 update broke booting from SATA devices that was working in 1.11.   

It just showed an EFI interactive screen with BLK2 referencing the SATA virtual device type.  It then just goes to a shell prompt which I have no idea what to type to get it to leverage the BLK2 (Sata interface).

By changing to IDE device, then boot proceeded normally again.

pvizeli, 2018-10-29:
I reupload the file @cooperj 

I use also IDE but I'll test it with SATA tonight. I don't change things on SATA driver, only a kernel update to 4.14.78 was performed

MwC-Trexx, 2018-10-29:
If somehow the filesystem within HASSO was damaged, are there any options for repairing it?

pvizeli, 2018-10-29:
I most case, the docker overlayfs2 as broken. They are on a data portion with ext4. I see that problem also on a test system but I was never be able to reproduce it. Not sure if that is a docker 18.03 problem and is fixed with 18.09.

You can see that with `journalctl -f` and there should be a message like `ERROR: readlink /var/lib/docker/overlay2: invalid argument`. You can fix that with `docker pull homeassistant/amd64-hassio-supervisor:latest`

MwC-Trexx, 2018-10-29:
No luck.  any way to get to the config files, so I can copy out changes I had made?

closed 2019-04-23 by stale[bot]