home-assistant/operating-system #238: PiFace does not work

RikVerstijnen, 2018-11-02:
PiFace was working well in combination with ResinOS, but I can't get it to working after upgrading to HassOS.

config.txt:
```
kernel=u-boot.bin
disable_splash=1
dtparam=audio=on
dtparam=spi=on
```

Log:
```
Error during setup of component rpi_pfio
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/setup.py", line 148, in _async_setup_component
    component.setup, hass, processed_config)  # type: ignore
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/rpi_pfio.py", line 25, in setup
    pifacedigital = PFIO.PiFaceDigital()
  File "/usr/local/lib/python3.6/site-packages/pifacedigitalio/core.py", line 82, in __init__
    self.init_board()
  File "/usr/local/lib/python3.6/site-packages/pifacedigitalio/core.py", line 107, in init_board
    h=self.hardware_addr, b=self.bus, c=self.chip_select))
pifacedigitalio.core.NoPiFaceDigitalDetectedError: No PiFace Digital board detected (hardware_addr=0, bus=0, chip_select=0).
```

pvizeli, 2018-11-05:
https://github.com/home-assistant/hassos/blob/dev/Documentation/boards/raspberrypi.md

RikVerstijnen, 2018-11-25:
Looks like [this PR](https://github.com/home-assistant/home-assistant/pull/18101) has solved my issues. I will close this one.

closed 2018-11-25 by RikVerstijnen