home-assistant/operating-system #246: Hassio does not start.

azizBayouni, 2018-11-12:
I'm trying to install HomeAssistant on my Raspberry PI 3 using this method ([https://www.home-assistant.io/getting-started/](url) ) but it never start and just keep showing (Preparing Hass.io) for 24 hours so far. I connected my PI with HDMI to see the progress. IT shows that it stuck at this ( [https://www.dropbox.com/s/3c5xflagy2f5gma/20181111_235144.jpg?dl=0](url)  ). Can you help me understanding the problem ? 

Thank you very much !

mkh595, 2018-11-19:
I have same issue. I tried install _hassos_rpi3-1.11.img_ and also _hassos_rpi3-1.12.img_ but it hangs after 28 seconds, see picture

[image]

pvizeli, 2018-11-19:
Login as root and show the logs with `su logs`

mkh595, 2018-11-19:
Log says
```The HTTP request failed with the error: Get http://hassio/supervisor/logs: dial tcp 172.30.32.2:80: connect: connection refused```

mkh595, 2018-11-19:
I restarted yesterday two RPI's via Hass.io>System>Reboot and both never started again. Then I tried new installation of HA from image but without success (picture above). To get my productive HA configuration working I copied SD card from test HA via HDDRawCopy, then restore snapshot, it started HA but someting is still wrong because sensor ```now().strftime('%A %d.%m.%Y %H:%M')``` shows now 19.11.2018 ```Monday 05.11.2018 20:37```

juandelaoliva, 2018-11-23:
I have the same problem with a Raspberry PI 3 and [Raspberry Pi 3 Model B and B+ 32bit](https://github.com/home-assistant/hassos/releases/download/1.13/hassos_rpi3-1.13.img.gz) image obtaining an picture similar to the previous one (for more than 14 hours). 
I have tried making the directory \CONFIG\network\ including the file my-network using a [HassOS howto](https://github.com/home-assistant/hassos/blob/dev/Documentation/network.md) sample, but I get the same result.

Thank you very much!

mkh595, 2018-11-23:
I tried image hassos_rpi3-1.13.img.gz and timesyncd.conf and instalation now works.
Thank you Pascal

ajdezigns, 2019-01-10:
Can someone please loop me in on what to do with the "timesyncd.conf" as in what it is and where i put/ alter it?? I'm having the same issues

jvanderneutstulen, 2019-01-10:
It is explained in [https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md](url). You may also want to try the new 2.5 image, it should use hardware random device, which will improve https performance at boot time quite a bit.

ajdezigns, 2019-01-10:
I have attempted the new image 2.5 it errors out at the same place with the same issue. I just tried you're the instructions you provided with 1.3 no luck going to try with 2.5 and see what happens. really sux i wanted to give my home assistant a bit more power then a pi 2 it's been using

ajdezigns, 2019-01-10:
so just tried that and no go here is a break down of my USB stick 
[image]

ajdezigns, 2019-01-10:
[image]

jvanderneutstulen, 2019-01-10:
when you login to the host system (first 'root' and then 'login'), can you
1. ping the dns servers?
2. ping the time servers?
3. what is the system's date/time?

ajdezigns, 2019-01-10:
That’s the thing it just keeps looping the text in the photo I posted it never gives me the opportunity to do anything

jvanderneutstulen, 2019-01-11:
I see a login prompt at the top in your photo, so you should be able to login locally using a keyboard. That should even work without any network. Also you might need to check your power supply or micro-usb cable, because you have 'Voltage normalized' message on the console. This indicates your pi has detected a low voltage condition, which may cause onboard devices to fail.

GPhilo, 2019-01-14:
I'm facing exactly the same problem, and I'm also out of ideas. I flashed the 1.13 image from the getting started and connected the RPi directly to the network via Ethernet. I can ping it, but no matter how long I'll wait, nothing ever shows up on port 8123. I even tried a different Raspberry to check if there's some hw failure but the issue is the same. The log from the RPi HDMI is useless (loops on a bunch of docker messages) and I can't get anything else from anywhere..  Honestly, I have no clue where to go from here.

mach7, 2019-03-18:
I'm having the same issues. Haven't been able to get my PI going either.

MTrab, 2019-04-16:
Mine worked for a few days and now it's just looping the gibberish docker text on HDMI - Hassio never comes backup.

hassos_supervisor starts, crashes, dissapears, reappears and then over again.