home-assistant/operating-system #251: something like cron?

pjv, 2018-11-20:
I've noticed that when wifi gets interrupted, my Pi with hassos does not seem to be trying to re-connect. That seems kind of bad for a home automation hub.

Anyway, I thought I would hack up a little script to ping my router every minute and reconnect to wifi on a ping fail. I'm hoping that `nmcli con up id hassos-network` (that last being the id i used in my connection config) will do the trick, but I don't know how to get that script to run periodically in hassos. Is there some equivalent of cron running?

pvizeli, 2018-11-20:
HassOS is only a platform for docker. Every logic needs to be inside a container, and they provide things like a cronjob.

Maybe we need to fix that on NetworkManager?

jvanderneutstulen, 2018-12-31:
NetworkManager does just that. I have tested this with a hotspot on my phone. After turning it on, NetworkManager connects within seconds. So I think your wifi is broken in such a way that NetworkManager stops reconnecting. Looking at [https://developer.gnome.org/NetworkManager/stable/settings-connection.html](url), I see two settings that might help you, `auth-retries` and `autoconnect-retries`, set them both to 0 in `[connection]` part of the configuration.

closed 2019-04-23 by stale[bot]