home-assistant/operating-system #253: Buildroot source for rpi-wifi-firmware - Error 404

sholzmayer, 2018-11-20:
Since a few days the hassOS build is no longer working:

```
--2018-11-19 15:51:57--  http://sources.buildroot.net/rpi-wifi-firmware-86e88fbf0345da49555d0ec34c80b4fbae7d0cd3.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.25.210.19, 104.25.211.19, 2606:4700:20::6819:d313, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.25.210.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-11-19 15:51:57 ERROR 404: Not Found.

make: *** [/build/buildroot/output/build/rpi-wifi-firmware-86e88fbf0345da49555d0ec34c80b4fbae7d0cd3/.stamp_downloaded] Error 1
package/pkg-generic.mk:135: recipe for target '/build/buildroot/output/build/rpi-wifi-firmware-86e88fbf0345da49555d0ec34c80b4fbae7d0cd3/.stamp_downloaded' failed
make: Leaving directory '/build/buildroot'
tee: build-log.txt: I/O error
ERROR: Job failed: exit code 1
```

After a quick look into the buildroot sources (`buildroot.com: /buildroot/package/rpi-wifi-firmware/rpi-wifi-firmware.hash`) i see a different (IMHO a newer) version of this package: 

```
# Locally calculated
sha256 155ebd5f08b819e0ce4e1950fcc972b2086cee3c16d36aba348beba1910c1fd2  rpi-wifi-firmware-8c1e2bff1da9850f68efcfff3da5d939ec27a2ee.tar.gz
sha256 b16056fc91b82a0e3e8de8f86c2dac98201aa9dc3cbd33e8d38f1b087fcec30d  LICENCE.broadcom_bcm43xx
```

The corresponding package-source file for that hash is downloadable: `http://sources.buildroot.net/rpi-wifi-firmware/rpi-wifi-firmware-8c1e2bff1da9850f68efcfff3da5d939ec27a2ee.tar.gz`.

I'd like to update the `0010-rpi-use-latest-wifi-driver.patch` to this newer (and downloadable) version. By doing this just in the patch-file, I still get an error due to the failing download of the buildroot package (because the patch is only applied after downloading and installing the not available package). So sadly it never reaches step 2, the patching...

My proposal would be to replace the `/buildroot/package/rpi-wifi-firmware/rpi-wifi-firmware.mk` with the newer official version of buildroot and remove the patch for that package. What do you guys say?

pvizeli, 2018-11-20:
We should use this one:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/brcm

That is the new and up to date source for blobs. I think the problem on https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm could be, that the commit was removed they we based on?

pvizeli, 2018-11-20:
The idea of buildroot-patches is that you can all time get a vanilla buildroot and apply the patches to have a working source for HassOS.

sholzmayer, 2018-11-21:
> The idea of buildroot-patches is that you can all time get a vanilla buildroot and apply the patches to have a working source for HassOS.

This is a good point. So I updated my PR, and I'm only replacing the rpi-wifi firmware with the current version of the buildroot-project. Afterwards it gets replaced with your patch and it shouldn't have any side effects to the current system, except that the build is not exiting with error 404. 

If anybody wanna get the current vanilla buildroot, this should be possible now...

pvizeli, 2018-11-26:
I think now you know why we need this: https://github.com/home-assistant/hassos/pull/258

Any way you build env is broken because I have no problem to build it.

closed 2018-11-26 by pvizeli