home-assistant/operating-system #257: No connection with bridged network

naamah75, 2018-11-25:
Using NAT I've no problem and with proper port forwarding all works very fine, but I would like to put it to work on bridge mode and in this way my host PC and all other devices on the network can't connect to the HASSos!
On virtualbox I simply select the bridge mode and the select the NIC card, but nothing … it doesn’t connect.
Viirtualbox have no error and tell me that the network is connected.

Tried to load a lubuntu iso instead of the HASSos image, and the bridge network works… so the problem is related to the HASSos system.

I would want to use the bridge mode beacuse I’ve a problem on connecting the Moquitto broker addon, it seems that won’t work through the NAT, even also this issue is very strange, beacuse all other server (including FTP) works through NAT.

guibulator, 2019-02-08:
For what its worth, I was having issue with bridge mode and it turned out to be working only when I explicitly set an ipv4 address with ipv4.method auto. Refer to this video [migrate hass.io from raspberry to virtual machine at about 8m10s](https://www.youtube.com/watch?v=vnie-PJ87Eg)
It worked for me.

naamah75, 2019-02-08:
Already do it what you worked for your case, but with no luck.

closed 2019-04-23 by stale[bot]

Freyert, 2022-03-13:
```console
login
nmcli
```
[image]

```console
cd /tmp
systemctl > f.txt
vi f.txt
```

[image]