home-assistant/operating-system #270: Hassos NTP feature

pergolafabio, 2018-11-30:
i see in hassos 1.13 a new NTP feature? what is this about?
i have an issue, when i reboot Hassos1.13 ,  (vmdk) that the timezone is 1 hour behind, because my timezone is ETA +1 
so all my sensors are not working in the first hour, or i just need to restart the HASS.io (restart config) just after the first  reboot of hassos , then its also solved... then my sensors are correct :)

so how does this NTP feature work?

thnx

ascillato, 2018-11-30:
@pvizeli 

Hi,

Can NTP be used to sync time for other devices in the network using http://hassio.local:123 ?

pvizeli, 2018-12-01:
@ascillato that need to be an add-on. You can create an add-on for time sync server and make a PR on core repository 👍

tinysnake, 2019-02-22:
I have a problem to proper install hass.io because of the time is way off so that unable to pull repositories from docker. I think the ntp server is inaccessible from China.
for more information please [read this forum post](https://community.home-assistant.io/t/connection-refused-frontend-doesnt-load-fresh-install/72675)

I'm pretty much a noob to docker and linux stuff, so I think could you add a feature to set the ntp server from file, like the "config/network/my-network" file, thanks!

closed 2019-03-28 by pergolafabio