home-assistant/operating-system #279: OVA Unable to boot with VirtIO disk controller 1.13

bagobones, 2018-12-07:
Hangs at:

Waiting for root device PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd...

Same message in 2.3 preview

Booting with emulated SATA works fine.

VirtIO network drivers are provided so I think proper VirtIO storage should also be.

VirtIO storage has substantial performance befits over emulated storage controllers. The drivers are also commonly used across many Linux based virtualization platforms.

MwC-Trexx, 2018-12-18:
Also allows for live snapshot backups of HASSOS

bagobones, 2018-12-18:
I would go a step further and ask for the client tools as well but I would settle for full virtual HW support since it is so common among linux virtualization platforms.

joeroback, 2018-12-27:
virtio_blk kernel driver is in 1.13 at least... doesn't load no matter how many virtio disks i attach to VM (even after reboot, etc)..  looking to see if its easy to enable this..

bagobones, 2018-12-27:
It isn't completely clear to me how the bootstrapping process in HASSOS works, a change from previous versions is that even with the VirtIO driver loaded it DOES at least start the boot process now instead of getting hung on drive selection. I think there is a hand off at some point that is missing something to allow VirtIO to work. It might be something in the boot loader or some other component like UBOOT that doesn't have the VirtIO storage support.

bagobones, 2019-02-19:
Still hangs in the same spot on HASSOS 2.9 if I switch my controller from SATA to VirtIO.

I haven't tried rebuilding from scratch but unless something is set and saved at some point it shouldn't make a difference.

nazar-pc, 2023-01-01:
Still getting stuck for me with 9.4, SATA worked fine. Maybe worth reopening? I'm on Alpine Linux host, using Libvirtd 8.9.0 with Qemu 7.1.0.

nazar-pc, 2023-01-01:
In fact, virtio network also didn't work, had to choose e1000e instead to get past NTP sync