home-assistant/operating-system #282: TTY devices do not show up in /dev on Raspberry PI 3B+ 64-bit

bobvanderlinden, 2018-12-09:
I have used the 64-bit image of Hass.io image for Raspberry PI 3B+. I connected an [AEON USB-stick](https://aeotec.com/z-wave-usb-stick), which should show up as an TTY device.

In the hass.io hardware configuration it shows up like you'd expect:

[image]

However, whenever I check in the shell whether the device exist it does not show up:

```
core-ssh:~# ls /dev
fd       full     init     mqueue   null     ptmx     pts      random   shm      stderr   stdin    stdout   tty      urandom  zero
```

Am I doing something wrong, or is this related to the issue about Bluetooth being in the way for the 64-bit images?

jvanderneutstulen, 2019-01-02:
This is expected behaviour. The 'shell' is running in a different docker container, which has no (tty)-device passthrough configured.

bobvanderlinden, 2019-01-03:
Where or how can I access the /dev that homeassistant has access to in hassio?

jvanderneutstulen, 2019-01-03:
If you enable the debug ssh, see [https://developers.home-assistant.io/docs/en/hassio_debugging.html](url), or attach a monitor/keyboard and login as 'root' you can gain access to the base system and /dev.

bobvanderlinden, 2019-01-03:
Aah, alright that makes sense. Will give this a try. Thanks for the hint.

closed 2019-01-03 by bobvanderlinden