home-assistant/operating-system #286: Any HTTPS Connection Blocked For Components

erfind, 2018-12-17:
Greetings,

I'm having an issue described at: https://github.com/home-assistant/home-assistant/issues/19060

I believe this issue may reside more with HassOS itself.  Is anyone else having this issue? I tried to grab the important parts of the issue I started. I would be happy to answer any questions or try anything to resolve the issue with some guidance :).

------------
Copied from the previous issue within Home Assistant Issue tracker:
------------

"Description of problem:
Whenever attempting to integrate with Nest at our office location, I kept getting internal validation errors and noticed they had to do with HTTPS requests having errors. I also noticed that my DarkSky was having issues as well and dug around in the logs and found out that it was getting the same error about HTTPS connections.

I had these previously components/integrations working on a Raspberry Pi running HassOS at our office. I now have the HassOS instance running in a VirtualBox with Bridged networking, allowing promiscuous mode, and have no issue accessing the Home Assistant via a HTTPS/SSL secured connection. I have already tried different DNS services on our office router, rebooting the HassOS instance to allow it to grab the new DNS servers with no change in errors. I don't use PiHole which I have seen people have issues with in the past. As stated before I used a Raspberry Pi with the same configuration a few weeks ago with no issue on this network and there are no blocked ports on our router.

I have another instance of Home Assistant running on a Raspberry Pi at our house, with the same Nest and DarkSky configuration as is being used at the office and the Nest and DarkSky both function without any issue at our house.

Any assistance would be awesome! :)"

----
"Try it with wget -O - URLHERE"
----

"I used the terminal within Home Assistant and got the following results. I tried multiple ways to address Nest's website and kept getting errors:

➜ ~ hassio --version
hassio version 1.4.0
➜ ~ which wget
/usr/bin/wget
➜ ~ wget -O - https://nest.com
--2018-12-10 10:05:00-- https://nest.com/
Resolving nest.com... failed: Try again.
wget: unable to resolve host address 'nest.com'
➜ ~ wget -O - http://nest.com
--2018-12-10 10:05:14-- http://nest.com/
Resolving nest.com... failed: Try again.
wget: unable to resolve host address 'nest.com'
➜ ~ wget -O - https://www.nest.com
--2018-12-10 10:05:26-- https://www.nest.com/
Resolving www.nest.com... failed: Try again.
wget: unable to resolve host address 'www.nest.com'
➜ ~ wget -O - http://www.nest.com
--2018-12-10 10:05:39-- http://www.nest.com/
Resolving www.nest.com... failed: Try again.
wget: unable to resolve host address 'www.nest.com'
➜ ~ wget -O - www.nest.com
--2018-12-10 10:05:52-- http://www.nest.com/
Resolving www.nest.com... failed: Try again.
wget: unable to resolve host address 'www.nest.com'

"

ecsfang, 2018-12-21:
Hi!
I have similar problem. Since the last weeks I can't reach anything outside my LAN, but from outside and in still works fine. E.g. I can't get any DarkSky data (unable to connect to Dark Sky), can't login on Verisure server nor push notifications to Pushover. But I can read sensors and update files using Configurator from my phone when not at home.
I'm using HassIO on a Raspberry Pi 3+, and uses DuckDns - and my log is full of SSL-problems.
I have been running this setup for several months, but suddenly noticed that external data is no longer available.
From the commandline I can e.g. reach GitHub, and I can ping Google, and running wget as shown above (at least from the Configurator (execute shell command)) works as expected.
I have seen many others with similar problems, but still no solution in sight that has worked for me ... :(
Cheers!

jvanderneutstulen, 2018-12-22:
@erfind You have a DNS configuration problem. Check with 
`
core-ssh:~# nslookup nest.com

core-ssh:~# nslookup nest.com 8.8.8.8
`

erfind, 2018-12-22:
Thanks for your reply :-)

I have attached my screenshot. It does appear that you are correct regarding the DNS issue. However, I have no issues with other devices on my network with respect to DNS. I used to have this same config running in a raspberry pi on this same network with no issue, and have changed DNS servers multiple times on my router in the past to try to resolve this since switching it to the VirtualBox.  

As I am running HassOS in a VM I have found no way to edit settings that would alter the DNS settings for my instance of HassOS. Is there a way to do this with the image that is provided for installing HassOS in VirtualBox?

jvanderneutstulen, 2018-12-22:
You can use the normal configuration options. If you create an extra disk with label CONFIG and put a networkmanager config on it, you should be able to use the 'import from usb'. Or maybe even use a real usb stick with usb passthrough.

But look into the natdnsproxy and natdnshostresolver options of virtualbox ([http://www.virtualbox.org/manual/ch09.html#vboxmanage-modifyvm-networking-nat](url)) first.

erfind, 2018-12-22:
Doing this worked!!!!!!!! Thank you @jvanderneutstulen you led me the right direction :) 

https://github.com/home-assistant/hassos/blob/dev/Documentation/network.md

Using nmcli to set a static IPV4 address

Log into the HASSOS base system via a console:

Welcome to HassOS
Hassio login:
Login as root (no password needed)

At the hassio > prompt, type login (as instructed).

From here you will use the nmcli configuration tool.

# nmcli connection show will list the “HassOS default” connection in use.

# nmcli con show "HassOS default" will list all the properties of the connection.

# nmcli con edit "HassOS default" will put you in a position to edit the connection.

nmcli> print ipv4 will show you the ipv4 properties of this connection.

To add your static IP address (select 'yes' for manual method);

nmcli> set ipv4.addresses 192.168.100.10/24
Do you also want to set 'ipv4.method' to 'manual'? [yes]:
In addition I have found it is wise to set the dns server and the local gateway. For most home routers these will be the same address. If you are using Pi-Hole you can set the dns to that.

nmcli> set ipv4.dns 192.168.100.1
nmcli> set ipv4.gateway 192.168.100.1
nmcli> save
nmcli> quit
If you now view the default connection cat /etc/NetworkManager/system-connections/default you should see the method is manual and the address is set.

Doing a nmcli con reload does not always work so restart the VM.

closed 2018-12-22 by erfind