home-assistant/operating-system #292: HassOS VMDK in ESXi 6.5 no USB visible

Emacee, 2018-12-28:
After I switched to HassOS from running Hass.io on an Ubuntu VM (with USB passthrough) my USB device is not visible anymore. On Ubuntu it worked without installing any additional drivers. The USB device is a RFlink hub/gateway. I prefer HassOS because it has less overhead than Ubuntu. Is there anything I can try to get the USB working?

The VMware ESXi setting of the VM shows the USB connected:
[image]

In Home-Assistant no USB devices show. 
[image]

In the logs I can only see that Home-Assistant has trouble connecting to the USB device which seems logical as it doesn't see the USB device at all.

pvizeli, 2018-12-28:
What for a kernel module do you need for this QingHeng USB serial converter?

Emacee, 2018-12-29:
Happy to try to find it! How do I know which kernel is needed? Can I see it in the Ubuntu Hassio installation? I’ve no experience in this type of thing.

Emacee, 2018-12-29:
Seems like it is using the CH341 driver instead of the one you linked:
[image]

When running lsmod before plugging in the device (via ESXi) and after only shows ch341 being added.