home-assistant/operating-system #317: Orange Pi Prime: Relatively slow kernel loading time vs Armbian

deece, 2019-01-11:
It takes ~9 secs to load the kernel for HassOS, but <1second for Armbian, @deece to investigate why.

Armbian:
8781859 bytes read in 445 ms (18.8 MiB/s)
14243848 bytes read in 713 ms (19.1 MiB/s)

jvanderneutstulen, 2019-01-11:
The Hassos kernel has almost everything compiled in, so has probably more to do.

pvizeli, 2019-01-19:
Yes, all kernel modules they we anyway need on runtime is compiled in. That makes docker faster startup.

deece, 2019-01-19:
That's not the problem, the sizes are similar, but the throughput is an order of magnitude different

jvanderneutstulen, 2019-01-23:
This comment, https://github.com/madisongh/meta-tegra/issues/42#issuecomment-323536525, in [u-boot: eMMC transfer speed significantly slower than stock u-boot on Tegra186](https://github.com/madisongh/meta-tegra/issues/42) is interesting.
Maybe running `mkfs.ext4` for the the kernel image with `-O ^extent` will help in our case.

pvizeli, 2019-01-31:
Can we do, it's only a small partition of 24mb

jvanderneutstulen, 2019-01-31:
Use `-O ^extent,^64bit` with our toolchain. Didn't notice any difference on a rpi3 with sdcard though.

deece, 2019-02-01:
I'm currently travelling,. But will investigate when I get home.

PR #338 (2019-02-03): Build: speedup uboot kernel load
Fix: #317
closed 2019-02-03 by pvizeli