home-assistant/operating-system #322: Jan 16 18:00:08 hassio hassos-supervisor[11018]: 19-01-16 18:00:08 ERROR (MainThread) [asyncio] Unclosed client session Jan 16 18:00:08 hassio hassos-supervisor[11018]: client_session: <aiohttp.client.ClientSession object at 0x750762f0>

snelweg, 2019-01-16:
/usr/sbin # systemctl status hassos-supervisor
● hassos-supervisor.service - HassOS supervisor
   Loaded: loaded (/usr/lib/systemd/system/hassos-supervisor.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-01-16 18:15:48 UTC; 28s ago
  Process: 23333 ExecStartPre=/usr/bin/docker stop hassos_supervisor (code=exited, status=0/SUCCESS)
 Main PID: 23342 (hassos-supervis)
    Tasks: 13 (limit: 2201)
   CGroup: /system.slice/hassos-supervisor.service
           ├─23342 /bin/sh /usr/sbin/hassos-supervisor
           └─23582 docker run --name hassos_supervisor --security-opt apparmor=hassio-supervisor -v /var/run/docker.sock:/var/run/docker.sock -v /var/run/dbus:/var/run/dbus -v /etc/machine-id:/etc/machine-id:ro -v /mnt/data/supervisor:/data -e SUPERVISOR_SHARE=/mnt/data/supervisor -e SUPERVISOR_NAME=hassos_supervisor -e HOMEASSISTANT_REPOSITORY=homeassistant/raspberrypi3-homeassistant homeassistant/armhf-hassio-supervisor

Jan 16 18:16:14 hassio hassos-supervisor[23342]:     addon_config = read_json_file(addon)
Jan 16 18:16:14 hassio hassos-supervisor[23342]:   File "/usr/local/lib/python3.7/site-packages/hassio/utils/json.py", line 21, in read_json_file
Jan 16 18:16:14 hassio hassos-supervisor[23342]:     return json.loads(cfile.read())
Jan 16 18:16:14 hassio hassos-supervisor[23342]:   File "/usr/local/lib/python3.7/codecs.py", line 322, in decode
Jan 16 18:16:14 hassio hassos-supervisor[23342]:     (result, consumed) = self._buffer_decode(data, self.errors, final)
Jan 16 18:16:14 hassio hassos-supervisor[23342]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
Jan 16 18:16:14 hassio hassos-supervisor[23342]: 19-01-16 18:16:14 ERROR (MainThread) [asyncio] Unclosed client session
Jan 16 18:16:14 hassio hassos-supervisor[23342]: client_session: <aiohttp.client.ClientSession object at 0x7510f270>
Jan 16 18:16:14 hassio hassos-supervisor[23342]: 19-01-16 18:16:14 ERROR (MainThread) [asyncio] Unclosed client session
Jan 16 18:16:14 hassio hassos-supervisor[23342]: client_session: <aiohttp.client.ClientSession object at 0x7510f390>

pvizeli, 2019-01-17:
Do you manually modify the file? Looks like they are UTF-16 not UTF-8

snelweg, 2019-01-18:
No, I did not change anything. This was a working 0.8.5 version, upgraded to 0.8.5.1 and then HA would not start again.
But isn't the real problem that aiohttp has left some unclosed session??? That error had been around for years now, apparently not fixed :(

pvizeli, 2019-01-18:
There is a fix they help for UTF-8 charset problem on Hass.io repository.

closed 2019-01-18 by pvizeli