home-assistant/operating-system #328: Add support for QinHeng USB2.0-Serial (aka CH34x)

mihalski, 2019-01-20:
Any chance of adding support for  QinHeng USB2.0-Serial (aka CH34x)? Currently the drivers seem to be missing and no ttyUSB device is created.
```
[78720.358964] usb 2-2.2: new full-speed USB device number 5 using uhci_hcd    
[78720.483119] usb 2-2.2: New USB device found, idVendor=1a86, idProduct=7523  
[78720.487625] usb 2-2.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[78720.488769] usb 2-2.2: Product: USB2.0-Serial  
```

Thanks for all your awesome work!

mihalski, 2019-01-21:
Awesome! Ironically the last dev version I ran was 2.3 and then when rebuilding my VMware hassio I went back to 1.13.

Is there yet a way to upgrade to the dev version in place or do I need to create a new instance and migrate the settings?

pvizeli, 2019-01-21:
You can move to the beta channel on Hass.io Panel and reload the supervisor. After that, you can update to RC (2.6).

mihalski, 2019-02-04:
Requires a addon restart to be acknowledged by esphomelibyaml addon (unlike for example Silabs and FTDI) but fully functional.

closed 2019-02-04 by mihalski