home-assistant/operating-system #344: HassOS 2.8 stuck at boot on tinker board

mbo18, 2019-02-04:
Hello,

Since the 2.8 release, I have an issue: sometimes it boots, sometimes not when ask to reboot from the Hassio panel. When it does not want to boot I need to unplug/replug the power cable and then it works again. I don't really know if this issue has been introduced with release 2.8 or a release before but I did not encounter it before

It is stuck at "Started Update UTMP about System Runlevel Changes." (just before "Welcome to HassOS" when it starts normally).

I have a tinker board S and HassOS runs from the eMMC.

[image]

cybergrimes, 2019-02-04:
I was just writing up an issue for the same problem which I believe it began with release 2.7-- using the reboot button from the Hassio system panel never works, I have to power cycle at the device.
I downgraded to both 2.5 and 2.6 this morning and am unable to reproduce the issue on those releases

pvizeli, 2019-02-05:
Hmm, that's funny because of for tinker was no changes on 2.7 and 2.8

mbo18, 2019-02-05:
It seems that this affect also HassOS running on other board
https://community.home-assistant.io/t/hassos-2-7-problems-after-update/96629
https://community.home-assistant.io/t/hassio-wont-start-failed-to-start-docker-application-container-engine/95407

cybergrimes, 2019-02-05:
I know the user makai at the end of [this thread](https://community.home-assistant.io/t/hassio-wont-start-failed-to-start-docker-application-container-engine/95407) is on a Tinker Board too

mbo18, 2019-02-05:
Yep it's me ;)

Maybe this PR (introduced in 2.7): https://github.com/home-assistant/hassos/pull/318

alex7614, 2019-02-08:
Hi

Same problem for me TinkerBoard S on eMMC 

2.6 -> It works
2.7 (update or fresh install) -> no boot
2.8 (update or fresh install) -> no boot

pvizeli, 2019-02-08:
I have no eMMC. Can someone test it for me?

alex7614, 2019-02-09:
I can test but how update to the specific version ?

closed 2019-02-09 by pvizeli

mbo18, 2019-02-09:
I'm currently building it. I will test it today or tomorrow.

@alex7614 to build it you need:
- Ubuntu 18.0.4 with docker-ce installed
- clone the branch: git clone -b tinker-kernel-4-19 https://github.com/home-assistant/hassos.git
- Follow this: https://github.com/home-assistant/hassos/blob/tinker-kernel-4-19/Documentation/getting_started_development.md
-> make tinker

mbo18, 2019-02-09:
@pvizeli I did the test:
First reboot did not worked (was stuck like on my picture) so I was thinking that the issue remained. But all my other attempts have worked as expected. So I think that this issue will be resolved when HassOS 3.0 will be released.
Thanks for all the work done

cybergrimes, 2019-02-11:
Looks good here on release 2.9

mizrachiran, 2019-03-12:
I tried setting up a new Tinkerboard, it didn't work for all versions (including v3.0).
The only version that worked for me is 2.6.

ndarilek, 2019-05-13:
Same here, 3.1 refuses to boot.

PR #349 (2019-02-09): Tinker: update kernel 4.19
fix: #344