home-assistant/operating-system #346: Can't update to 2.8

jamesking420, 2019-02-05:
WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.8/hassos_rpi3-2.8.raucb: Cannot connect to host github-production-release-asset-2e65be.s3.amazonaws.com:443 ssl:None [Try again]

jamesking420, 2019-02-05:
I've previously been able to successfully upgrade to 1.13 from 1.12 (for example).

jamesking420, 2019-02-05:
When connected to the same light network I can download the image from the URL fine (to my phone.)

jamesking420, 2019-02-05:
This has been happening for a couple of days, although the final sentence about amazonaws was new tonight. Previously it gave no explanation.

jvanderneutstulen, 2019-02-06:
Can verify your raspi has the correct time? An invalid time will cause ssl connections to fail.

jamesking420, 2019-02-06:
It is showing the correct time for me here in the UK.

pvizeli, 2019-02-07:
Strange. Do you have any special network configuration? If you speak about CLI, is that from SSH add-on or local console?

jamesking420, 2019-02-07:
No particular special network add on. Pi is wired through to my router.

The SSH add on.

cknieriem, 2019-02-08:
I am currently running HA on an RP3B+ running  HASSOS.  On the Host System page it states that I am at HassOS 1.13 and the UPDATE option is visible.  When I try to update, I see the following error in the logs: 19-02-08 21:09:18 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.8/hassos_rpi3-2.8.raucb: Response payload is not completed.  I am not running PiHole.  I am running DuckDNS.

tomlut, 2019-02-10:
Seeing the same issue. No special network configuration (DuckDNS) and HA time is correct.

tomlut, 2019-02-11:
Update button is now working for me.

cknieriem, 2019-02-11:
I am still unable to update with the same error as my previous post.

jamesking420, 2019-02-11:
Yep, I'm also still unable to update.

Happy to provide more detailed logs if someone can tell me how to get them?

pvizeli, 2019-02-11:
Which country?

cknieriem, 2019-02-11:
US

pvizeli, 2019-02-11:
Is the time and date correct?

cknieriem, 2019-02-11:
Time and date are correct.

jamesking420, 2019-02-11:
I'm in the UK, correct time and date, same issue.

jamesking420, 2019-02-11:
I have my Pi set up with dynamic IP address, which is then given by the router, using a DHCP reservation.  So it always gets the same IP address but it is not a static IP address.

cknieriem, 2019-02-11:
I have configured a true static IP address on the PI.

pvizeli, 2019-02-11:
The link work inside your networks: https://github.com/home-assistant/hassos/releases/download/2.8/hassos_rpi3-2.8.raucb ?

cknieriem, 2019-02-11:
I can connect to the GitHub site from inside my network.  I can also connect to the other Add-On Github repositories within HA

jamesking420, 2019-02-11:
Plenty of space on my SD card. 

I guess I'll have to give that a try. Might wait until my pi sees 2.9 and then see if that works.

cknieriem, 2019-02-11:
So I create a CONFIG folder on a USB stick and put the contents of the download in the CONFIG folder?

cknieriem, 2019-02-11:
I have 22 GB free space on my SD card for HA.

jamesking420, 2019-02-11:
Got a different error this time:

19-02-11 20:16:26 INFO (MainThread) [hassio.homeassistant] Updated Home Assistant API token
19-02-11 20:33:50 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.8/hassos_rpi3-2.8.raucb
19-02-11 20:38:50 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.8/hassos_rpi3-2.8.raucb: 
19-02-11 20:38:50 ERROR (MainThread) [aiohttp.server] Unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 447, in start
    await resp.prepare(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_response.py", line 353, in prepare
    return await self._start(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_response.py", line 667, in _start
    return await super()._start(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_response.py", line 410, in _start
    await writer.write_headers(status_line, headers)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/http_writer.py", line 112, in write_headers
    self._write(buf)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/http_writer.py", line 67, in _write
    raise ConnectionResetError('Cannot write to closing transport')
ConnectionResetError: Cannot write to closing transport

BasSmeets, 2019-02-13:
Same issue.

jamesking420, 2019-02-15:
So whgat's the correct method for updating via USB?

Create a USB stick where rhe only partition is called CONFIG and place the update file on that USB stick?

jamesking420, 2019-02-18:
Managed to do this follwoing some guidance here:
https://github.com/home-assistant/hassos/blob/dev/Documentation/configuration.md
•The hassos-*.raucb file is a firmware OTA update which will be installed. These can be found on on the release page.

It's as I said above - a USB with one partition called "CONFIG".  I created the partition, shut down Hassio host (from the web interface for Home Assistant), plugged in my USB stick, restarted my Pi, and then clicked thge "Update from USB" button on the web interface when it was back up and running.  It took a while to actually do anything, but after a while it rebooted on the up-to-date Hass.os.

kajmaj, 2019-03-04:
The same issue here while updating from 2.8 to 2.10, all previous updates were problemless.

```
19-03-04 08:14:12 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.10/hassos_rpi3-2.10.raucb
19-03-04 08:19:13 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.10/hassos_rpi3-2.10.raucb: 
19-03-04 08:19:18 ERROR (MainThread) [asyncio] SSL error errno:1 reason: KRB5_S_INIT
protocol: <uvloop.loop.SSLProtocol object at 0x74c94c10>
transport: <TCPTransport closed=False reading=False 0x71a54600>
Traceback (most recent call last):
  File "uvloop/sslproto.pyx", line 504, in uvloop.loop.SSLProtocol.data_received
  File "uvloop/sslproto.pyx", line 204, in uvloop.loop._SSLPipe.feed_ssldata
  File "uvloop/sslproto.pyx", line 189, in uvloop.loop._SSLPipe.feed_ssldata
  File "/usr/local/lib/python3.7/ssl.py", line 767, in unwrap
    return self._sslobj.shutdown()
ssl.SSLError: [SSL: KRB5_S_INIT] application data after close notify (_ssl.c:2605)
```
Any advice except USB option?

GaryOkie, 2019-03-15:
And the same HassOS update issue for me too  (errors identical to above).   I confirmed date and time were accurate.   SD card /home use is 10% (of 32GB).  

Had no problem updating HA to 89.2 a few minutes earlier from Github, so network connection should be fine.   

This is a remote system that I maintain and using the USB update option is not possible unless I drive 100+ miles. 

Any advice other than USB option?

mkh595, 2019-04-18:
Same issue when trying update from 1.13 to 2.11
```
19-04-18 07:26:36 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.11/hassos_rpi3-2.11.raucb
19-04-18 07:31:37 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.11/hassos_rpi3-2.11.raucb: 
19-04-18 07:31:37 ERROR (MainThread) [asyncio] Fatal error on transport
protocol: <uvloop.loop.SSLProtocol object at 0x756338c8>
transport: <TCPTransport closed=False reading=False 0x71a269e0>
Traceback (most recent call last):
  File "uvloop/sslproto.pyx", line 574, in uvloop.loop.SSLProtocol._do_shutdown
  File "/usr/local/lib/python3.7/ssl.py", line 767, in unwrap
    return self._sslobj.shutdown()
ssl.SSLError: [SSL: KRB5_S_INIT] application data after close notify (_ssl.c:2609)
```

jamesking420, 2019-06-17:
Still happening with more recent versions of Hass.os

closed 2019-08-23 by stale[bot]

allmoney, 2019-12-20:
I have error
19-12-20 14:58:19 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.7/hassos_rpi4-3.7.raucb: 

Cant update :(

billclerico, 2020-01-02:
I have the same issue

20-01-02 04:06:42 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/3.7/hassos_rpi3-3.7.raucb
20-01-02 04:11:42 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.7/hassos_rpi3-3.7.raucb:

hadrianl, 2020-01-17:
same error!!!

cabe1214, 2020-01-28:
Same error

`20-01-28 19:48:43 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/3.8/hassos_rpi4-3.8.raucb: Cannot connect to host github.com:443 ssl:None [Try again]`

JSchoeck, 2020-05-21:
> I tried and I can download the file with wget from the terminal.
Can you start the update via SSH if you have the file on the HassOS filesystem?

ianhyzy, 2021-02-28:
I'm also having this issue