home-assistant/operating-system #350: Can't start web after _ho reboot_ on Pi3B

KKnishev, 2019-02-09:
can't start web after _ho reboot_ on Pi3B
hostname: hassio
kernel: 4.14.81-v7
operating_system: HassOS 2.8
HELP PLEASE!!!

Log not from my system, but the same:
```
[   30.869545] kauditd_printk_skb: 26 callbacks suppressed
[   30.869556] audit: type=1400 audit(1545951108.100:65): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor" pid=3046 comm="apparmor_parser"
[   30.894461] audit: type=1400 audit(1545951108.100:65): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=3046 comm="apparmor_parser"
[   30.915528] audit: type=1400 audit(1545951108.100:65): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/git" pid=3046 comm="apparmor_parser"
[   30.936263] audit: type=1400 audit(1545951108.100:65): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="hassio-supervisor///usr/bin/socat" pid=3046 comm="apparmor_parser"
[   30.957165] audit: type=1300 audit(1545951108.100:65): arch=c00000b7 syscall=64 success=yes exit=68940 a0=6 a1=ffffb99b0010 a2=10d4c a3=0 items=0 ppid=3045 pid=3046 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="apparmor_parser" exe="/usr/sbin/apparmor_parser" subj==unconfined key=(null)
[   30.987601] audit: type=1327 audit(1545951108.100:65): proctitle=61707061726D6F725F706172736572002D72002D57002D4C002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F6361636865002F6D6E742F646174612F73757065727669736F722F61707061726D6F722F68617373696F2D73757065727669736F72
```

KKnishev, 2019-02-09:
[image]

pvizeli, 2019-02-11:
Log into host and post the logs of `su logs`, `ha logs`

KKnishev, 2019-02-11:
Thanks for support @pvizeli, sorry for screenshots, don't know how to save files from Pi3.
su logs at screen
ha log is empty, it's down as I see
[image]

jvanderneutstulen, 2019-02-11:
Is your HA config valid  (`ha check`) ?

KKnishev, 2019-02-12:
hassio > ha check
**ERRO**[0000] Unexpected server response
Unexpected server response
hassio >

pvizeli, 2019-02-12:
- `ha stop`
- `ha start`

KKnishev, 2019-02-12:
hassio > ha stop
Error:

hassio > ha start
**ERRO**[0000] Unexpected server response
Unexpected server response

KKnishev, 2019-02-12:
Сan this be related to this topic (AppArmor profile):
https://developers.home-assistant.io/docs/en/hassio_addon_security.html

pvizeli, 2019-02-12:
do you see anything inside supervisor logs after you execute these commands? Home Assistant has no AppArmor profile.

KKnishev, 2019-02-12:
Now yes!
[image]

KKnishev, 2019-02-12:
How and where find position 6494 ??
Here is a similar [issue](https://community.home-assistant.io/t/strange-errors-not-sure-how-to-troubleshoot/47969)

KKnishev, 2019-02-13:
Unfortunately I could not restore it, reinstalled with backup. Thanks for support. Closed.

closed 2019-02-13 by KKnishev