home-assistant/operating-system #351: RockPro64 Support

s4nch3z, 2019-02-09:
@pvizeli,
Could you please outline what steps it would take to support RockPro64 board?
I'm willing to help with this feature development.

Thank you.

rbray89, 2019-02-10:
1) Create a new config in buildroot-external/configs. Easiest to copy a similar device and change the contents. eg. `buildroot-external/configs/rock_pro_64_defconfig`
    1) Tweak the rest of the defconfig to point to valid docker images, sources, etc.
2) Create a new series of board folders and files for use with this specific board. eg. `buildroot-external/board/pine64/rockpro64/`
3) Determine kernel/uboot support & patches. Set version/source in the defconfig file, and patches should be placed in the board directory under patches/$COMPONENT' eg. `buildroot-external/board/pine64/rockpro64/patches/linux` If patches are needed, add the patches directory to the BR2_GLOBAL_PATCH_DIR variable in the defconfig.
4) Populate meta fiile. eg. `buildroot-external/board/pine64/rockpro64/meta`. This file contains some common info needed by the build system.
```
BOARD_ID=odroid-c2 <-ID used by docker and build system (no spaces)
BOARD_NAME="Hardkernel Odroid-C2" <- Friendly name
CHASSIS=embedded <- embedded or vm
BOOTLOADER=uboot <- uboot or barebox
KERNEL_FILE=Image <- Kernel image format. Usually one of Image, zImage, or bzImage
BOOT_SYS=mbr <- Partitioning layout. spl, mbr or efi. This depends on what kind of firmware the chipset is running and how it boots. Right now we don't have great documentation on getting devices to boot. The best thing to do here, is look at the vendor's documentation (or other documentation from other embedded OS projects) to understand how the board boots and find the most similar boot mechanism. SPL is used if the board can use a GPT partitioned layout with an SPL image in a specific sector. MBR is used in the same circumstance, but it must be MBR partitioned. EFI is used for EFI boot where there is already a capable boot loader on-board.
BOOT_ENV_SIZE=0x2000 <- Should be aligned with u-boot env  size, but realistically just needs to be large enough for a couple variables. 
```
5)
    5) If this is an embedded board utilizing uboot and can also boot from multiple sources (USB, eMMC, SD, built-in flash, etc.) copy another board's uboot-boot.sh file eg. `buildroot-external/board/hardkernel/odroid-c2/uboot-boot.ush` and determine what needs to be updated. 
    5) If the board can only boot from one source but is uboot, copy the rpi boot script `buildroot-external/board/raspberrypi/uboot-boot.ush` and update accordingly.

6) Build & flash.

Notes)
* It is best if you have serial access to the board/device you are trying to develop on to be able to access early boot info (from uboot). Drivers for Ethernet, USB, and graphics may not exist, load, or simply work, so without serial the road will be long and arduous. 
* Getting the board to boot properly is the most difficult part. It's usually easiest to start with a known working build system (vendor's, armbian, etc.) to see how they lay out the filesystem, what versions of u-boot they run, kernel versions, etc, and start there, tweaking to later/mainline u-boot and kernel sources.
* Mainstream kernel is not required, but it is encouraged if it can be stably used with patching. Using third party (aka. vendor) kernels can be OK, but it is less preferred as changes to those kernels aren't always upstreammed, and worse,  could be hiding nasty surprises.

s4nch3z, 2019-02-12:
@rbray89, 
Thank you so much! Now it will take some time for me to process all this. I'll ping you here once I get somewhere.

closed 2019-04-23 by stale[bot]