home-assistant/operating-system #357: Bluetooth USB dongle (Broadcom) not detected...

Swiftnesses, 2019-02-13:
Hi,

I'm unable to get this detected under hardware, `lsusb` shows:

```
Bus 002 Device 002: ID 0e0f:0003
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 004: ID 0a5c:21ec
Bus 002 Device 003: ID 0e0f:0002
Bus 002 Device 001: ID 1d6b:0001
Bus 002 Device 005: ID 0658:0200
```

The device is `0a5c:21ec` and is listed here: https://github.com/winterheart/broadcom-bt-firmware

Is there anything I can do to get this working?

Swiftnesses, 2019-03-03:
I've now purchased and tried 3 BT dongles without luck...

I'm running ESXI 6.7 and the device is attached to the VM (shows in console on boot).

@pvizeli is there anything I can do to workaround this, I'm pretty desperate :/

Thanks in advance.

Swiftnesses, 2019-03-07:
Seems I need: BCM20702A1-0a5c-21e8.hcd

The device is actually BCM20702A0-0a5c-21e8.hcd, but it's a common issue on newer kernals that it actually wants to see BCM20702A**1**-0a5c-21e8.hcd in lib/firmware/bcrm rather than the expected BCM20702A**0**-0a5c-21e8.hcd

lpt2007, 2019-03-10:
Did you solve problem with usb dongle. I have same problem with [logitech dinovo mini](https://github.com/home-assistant/hassos/issues/373).

Swiftnesses, 2019-03-10:
Had no choice but to move to Debian 9, sadly...

CinoNY, 2019-03-20:
Wish I came across this a couple of days ago. I have ESXi 6.7 running on a NUC and no dice with the built-in BT. The BT dongle I just received didn't work either. I'm assuming your running a Debian 9 VM on your ESXi host? Really dont want to give up HassOS if I can but might have too :-(

SPCulhane, 2019-05-20:
@pvizeli Is this on your radar for HassOS? I would like to stay on HassOS if possible.

closed 2019-07-26 by stale[bot]

tybritten, 2020-06-19:
Just to add a 👍🏼 to this closed issue. I'm moving off HassOS to Debian for the same reason. I'd love the ability to inject other firmware and drivers via the CONFIG USB

SilverWx, 2020-09-06:
Please do something, I bought 3 dongle included wide used on linux pluggable BT4.0 without luck

sebi5361, 2021-01-12:
I am having a similar issue: I am willing to use an external USB Bluetooth dongle in place of the embedded Bluetooth device on my Raspberry Pi 3B+ running the latest version of Home Assistant Operating System (to extend the range).

`lsusb` shows it is plugged in and gives the standard `0a12:0001` ID corresponding to a CSR8510A chipset, a linux compatible Bluetooth 4.0 dongle.
`bluetoothctl` lists only the onboard BlueTooth device as a registered agent. The dongle is not listed.

Maybe the dongle drivers are missing from Home Assistant Operating System? Is there a way to add those drivers to the OS?

toreupfeet, 2021-01-27:
I'm having issues trying to get my Plugable Bluetooth 4.0 LE USB adapter to work with Debian 10 on VirtualBox. I had the `failed to load brcm/BCM20702A1-0a5c-21e8.hcd` error, which seems to have been resolved.

I followed the instructions provided here: [https://www.theoutpost.org/uncategorized/bluetooth-hci0-firmware-failed-to-load-brcm-bcm20702a1-0a5c-21e8-hcd-2/](url)

```
I fixed this by….
1, unplug bluetooth usb
cd /opt
wget https://s3.amazonaws.com/plugable/bin/fw-0a5c_21e8.hcd
mv fw-0a5c_21e8.hcd /lib/firmware/brcm/BCM20702A0-0a5c-21e8.hcd
cp /lib/firmware/brcm/BCM20702A0-0a5c-21e8.hcd /lib/firmware/brcm/BCM20702A1-0a5c-21e8.hcd

re-insert usb bluetooth device
```
I now have a new issue: `Bluetooth: hci0: BCM: Reset failed (-110)`

I have tried the USB 3 ports on the rear of my tower to no aval. It seemed to have worked for someone else, though. See here: https://askubuntu.com/questions/1250116/broadcom-usb-bluetooth-adapter-not-working