home-assistant/operating-system #358: Instructions for USB boot

Ayo91, 2019-02-17:
How can I boot the system using an external SSD?

rbray89, 2019-02-18:
What platform?

Ayo91, 2019-02-18:
Hass.io and HassOss

rbray89, 2019-02-18:
No, I mean what hardware platform.

Ayo91, 2019-02-18:
Ahh ok, sorry. Raspberry Pi 3B

rbray89, 2019-02-21:
Unfortunately we can't boot from most embedded devices using USB. There is a thread [https://github.com/home-assistant/hassos/issues/164#](here) That I see you've already found that documents the same use-case.

The RPi3B+ is special though, in that you CAN permanently enable USB boot with a FW configuration mechanism, but our boot-loader simply isn't setup to handle that case.

Ayo91, 2019-02-21:
ok, thanks. Is there any option to reduce the overload of the SD card and use together with an USB SSD in the Raspberry Pi 3B?

talondnb, 2019-03-02:
> Unfortunately we can't boot from most embedded devices using USB. There is a thread [https://github.com/home-assistant/hassos/issues/164#](here) That I see you've already found that documents the [...]

So are you suggesting there are no plans to get USB/SSD boot working?

closed 2019-05-08 by stale[bot]

tiagofreire-pt, 2019-05-10:
> The RPi3B+ is special though, in that you CAN permanently enable USB boot with a FW configuration mechanism, but our boot-loader simply isn't setup to handle that case.

That is only applicable on 3B, not 3B+, as the first one has a one time non reversible setting.