home-assistant/operating-system #364: SSH daemon has debug logging

robin13, 2019-02-22:
**Home Assistant release with the issue:**
Hassio: 0.88.0

**Component/platform:**

HassOS: 2.8
board: rpi3
SSH Server: 5.1

**Description of problem:**

In the ssh daemon configuration, debug logging is enabled:

```
LogLevel DEBUG
```
This results in this line being output every time one connects:

```
debug1: permanently_set_uid: 0/0
```
And this in turn breaks backuppc from being able to connect to make a backup and other ssh based automations.

Is it possible to set LogLevel to INFO?

bwolther, 2019-04-14:
Image: homeassistant/raspberrypi2-homeassistant
HassOS: 2.1
Home Assistant version: 0.91.3
Hass.io add-on: SSH server

I encountered this problem on the latest version when I tried to execute the following command from a non-hassio linux host ('hassio' is the name of the host on which I'm running hassio):
```
ssh hassio ls -ld /config/`
```
That gave me the following output:

```
debug1: permanently_set_uid: 0/0
drwxr-xr-x    7 root     root          4096 Apr 14 15:39 /config/
```
The first line ("debug1:...") appears to be extraneous debug output coming from the current installation's sshd server add-on.

I was able to somewhat work around that by redirecting stderr to /dev/null like this:
```
ssh hassio ls -ld /config/ 2>/dev/null
```

closed 2019-06-20 by stale[bot]

bwolther, 2019-08-19:
This problem still exists.