home-assistant/operating-system #367: Can't update to 2.10

jamesking420, 2019-02-24:
I'm now getting the following when trying to update from 2.9 to 2.10, which might be a more informative error message?

19-02-24 08:13:21 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.10/hassos_rpi3-2.10.raucb
19-02-24 08:18:22 WARNING (MainThread) [hassio.hassos] Can't fetch versions from https://github.com/home-assistant/hassos/releases/download/2.10/hassos_rpi3-2.10.raucb: 
19-02-24 08:18:22 ERROR (MainThread) [asyncio] SSL error errno:1 reason: KRB5_S_INIT
protocol: <uvloop.loop.SSLProtocol object at 0x72071e90>
transport: <TCPTransport closed=False reading=False 0x71991bd0>
Traceback (most recent call last):
  File "uvloop/sslproto.pyx", line 504, in uvloop.loop.SSLProtocol.data_received
  File "uvloop/sslproto.pyx", line 204, in uvloop.loop._SSLPipe.feed_ssldata
  File "uvloop/sslproto.pyx", line 189, in uvloop.loop._SSLPipe.feed_ssldata
  File "/usr/local/lib/python3.7/ssl.py", line 767, in unwrap
    return self._sslobj.shutdown()
ssl.SSLError: [SSL: KRB5_S_INIT] application data after close notify (_ssl.c:2605)

pvizeli, 2019-03-18:
Is the time correct?

jamesking420, 2019-03-18:
It was, yes.

closed 2019-03-18 by pvizeli