home-assistant/operating-system #368: Install hassos in docker container?

lpt2007, 2019-02-25:
How to install hassos in docker container?

tboyce021, 2019-04-04:
HassOS is an actual operating system, think Ubuntu, Windows, etc. It is meant to be a lightweight/minimal OS that works for Hass.io and can be updated through the Hass.io interface.

Hass.io is a set of Docker containers and software that manage the Home Assistant installation and add-ons.

If you already have an OS with Docker, you can just install Hass.io on it as mentioned above.

closed 2019-06-10 by stale[bot]

asemev, 2020-05-10:
Now with the latest announcement on May 9th 2020 "Deprecating Home Assistant Supervised on generic Linux", this topic could be relevant again. If it would be possible to install the whole HassOS inside a docker container (instead of a VM), it would be so much nicer... Any thoughts?

frenck, 2020-05-10:
@asemev That is not possible. It is not just a Docker container.

asemev, 2020-05-10:
@frenck - Thank you Frenck.. I am noob, so i dont really understand what do you mean by "not just a Docker container"... but thanks a lot for your answer.

Circuitsoft, 2021-11-14:
It is certainly possible to run Docker on an OS inside a Docker container, just needs to be `--privileged.` See [Docker can now run within Docker](https://www.docker.com/blog/docker-can-now-run-within-docker/). What's to prevent me from starting a Debian 10 docker image with `--privileged` and just unpack the whole OS image into it?

agners, 2021-11-18:
@Circuitsoft the system will pretty much fall apart: HAOS starts at a much lower level: It comes with a boot loader, with two root file system, with fallback logic in case one of the root file system happens to be corrupt/non-bootable. The root file system itself is a read-only squashfs. HAOS uses systemd as init system and it mounts additional partitions detected via udev. Those additional partitions are the data partitions where Docker is running on. The update system `rauc` also relies on hardware access to write file system images, so OS update won't work either.

While you might be able to unpack the squashfs main root file system, and get "something" going, it won't work by default for sure.

And it also defeats the purpose of HAOS: It's supposed to be a bare metal operating system. Supervised is a supported installation option, so what you are looking for is a Supervised installation.