home-assistant/operating-system #371: Unmount errors during HASSOS (2.10) Shutdown

MwC-Trexx, 2019-02-26:
During HASSOS shutdown, several unmount errors are being recorded>

**OS:** HASSOS 2.10
**Platform:** QEMU_x64
**HASSIO version:** 0.87.2

During HASSOS shutdown, the following errors are being being shown (see screenshots).

[image]

I wonder if this is what is causing some corruption of HASSOS VM's if there are frequent shutdown/reboots.

jvanderneutstulen, 2019-02-26:
Maybe unmounting fails because of corruption. Can you capture a more complete log using a serial console?

MwC-Trexx, 2019-02-26:
I will see what I can do.... since QNAP Virtualization Station is based around QEMU/KVM, it MIGHT be possible to do something at CLI level, but not sure. There definitely isn't a console "log" option as far as I am aware.

Based on the screen logs/error messages is seemed more like a code issue 

- Docker container unmoun: Invalid Argument.
- Overlay2 unmount: Invalid Argument

I am still a semi-novice in some of this space though.

jvanderneutstulen, 2019-02-27:
It is also possible to use a GUI, for example virt-manager. You can select the 'text console' you want to use. Just append `console=ttyS0,115200n8` to the `cmdline.txt`.

MwC-Trexx, 2019-03-01:
No such luck.  Qnap doesn't leverage virt-manager nor expose libvirtd externally.

pvizeli, 2019-04-05:
Should fix with 2.11?

closed 2019-06-11 by stale[bot]