home-assistant/operating-system #372: Support for DS3231 RTC module for the Raspberry Pi

jasiek, 2019-03-04:
Hey,

There are tons of modules like this one (https://www.ebay.co.uk/itm/DS3231-RTC-Real-Time-Clock-Module-for-Raspberry-Pi/192664694929) out there - I use it on all of my PIs for time keeping. Would it be difficult to enable i2c and load the appropriate module for this to work? I reckon there may need to be a need to run hwclock at some point to sync the time as well ...

rradar, 2019-03-25:
Uuuu...

I was just running in this whole. Switched to hassio in a true® offline environment and have now big troubles keeping time. My raspberry has a DS3231 RTC but hassio doesn't do anything with it. It just tries to reach the big brother (google) to get the time... but in a true® offline world big brother is not accessible...

rradar, 2019-05-07:
hassos doesn't support it for now. Just get rid it of it (install hassio on raspbian) and you have the freedom you are seeking (using the rtc module as expected).

avbfr, 2021-06-04:
Hello,

I sucessfully configured a DS3231 on hassos just by enabling i2c. The RTC works fine and can be read using hwclock command on hassos. 

I can manualy set System Time from RTC using _hwclock --hctosys_ but it would be interresting to do it automaticaly on startup. 
For now, If my HA is not connected to internet and shutdown system time won't be synchronised from RTC. 

Thanks !

agners, 2021-06-05:
From what I can tell the RTC driver is compiled in `CONFIG_RTC_DRV_DS1307`, so this should be a matter of correct overlays applied.

@avbfr can you share how you enabled I2C?

From what I can tell something like the following configuration in `config.txt` should do it:

```
dtoverlay=i2c-rtc,ds3231
```

See also: https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time

avbfr, 2021-06-05:
Hello @agners, 

Thanks for your answer. 
That's exactly what i have: 
_dtparam=i2c_vc=on_
_dtparam=i2c_arm=on
_dtoverlay=i2c-rtc,ds3231_

**My RTC is correctly configured and I can get time with hwclock command.** 

What is missing is a startup script that synchronise system clock from hardware clock (using hwclock -s) as also described in the link you provided for raspbian in the /etc/rc.local section (which does not exist in hassos).

As most of the OS is Read Only, It's not trivial on the user side to do it.

Thanks !

agners, 2021-06-05:
> What is missing is a startup script that synchronise system clock from hardware clock (using hwclock -s) as also described in the link you provided for raspbian in the _/etc/rc.local_ section (which does not exist in hassos).

That is not really necessary as the Linux kernel is automatically synchronizing the system clock with the first found rtc (`rtc0`). E.g. this is from ODROID-N2+ (Blue):

```
[    1.245792] rtc-pcf8563 0-0051: registered as rtc0
[    1.247787] rtc-pcf8563 0-0051: setting system clock to 2021-06-05T12:05:56 UTC (1622894756)
```

You should have a similar entry in the kernel log after boot.

> For now, everytime the system is down, the system clock is only synchronized from NTP (using datetimectl started by systemd) but not from RTC. 

Note that there are two daemons running to deal with time: `systemd-timedated` and `systemd-timesyncd`. They handle synchronization as well as storing the synchronized time in the hardware RTC.

By default we read the time from RTC but *still* delay boot up to 90s to wait for NTP synchronization. Unfortunately, a RTC gives you no guarantee that the stored time is correct... And we had several issue reports where the RTC was wrong, with successive failures to connect to TLS (https) servers because the certificates were no longer considered valid or valid in the future due to wrong date/time. See also #1360.

So after enabling RTC you should see this behavior (and this is the intended behavior):
- With Internet, the boot will be slightly delayed until time synchronization succeeded
- Without Internet, the boot will wait up to 90s to wait for time synchronization. If that is unsuccessful, boot should continue. The system should have the stored time from RTC.

avbfr, 2021-06-10:
Hello @agners. 
Thanks for your answer. 
I'm going to test this again.

codeclinic, 2023-03-24:
@agners Any update? Could you detail the steps to get the RTC module working?

agners, 2023-03-24:
RTC does work on HAOS by default using `config.txt` modifications.

This is not the platform to discuss how to do this. Please use the community forums, along with details what RTC/RPi you are using, and what you tried and what you are trying to archive. The community might be able to help. https://community.home-assistant.io/