home-assistant/operating-system #376: x server app as hassio addon?

lpt2007, 2019-03-13:
I wondering if it possible to make hassio addon to start x server app like chrome on x display of host (in my case lcd connected to raspberry pi 3 hdmi)?

lpt2007, 2019-03-14:
Probably there is no minimal x11 packages installed to start x display?

lpt2007, 2019-03-14:
I try buildroot rpi3 image from this [link](https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Buildroot.html) and touch screen and hdmi output is working.

@pvizeli can you add missing packages to make use of hdmi (x display) and usb touchscreen possible.

lpt2007, 2019-03-14:
I try buildroot rpi3 image from this [link](https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Buildroot.html) and touch screen and hdmi output is working.

Can you add missing packages to make use of hdmi (x display) and usb touchscreen possible

lpt2007, 2019-03-14:
Sorry I accidently close this

lpt2007, 2019-03-14:
I think package [xorg](https://git.busybox.net/buildroot/tree/package/xorg?id=8118da144991ab7004522b1e16f06675cc31b3ec) and [x11r7](https://git.busybox.net/buildroot/tree/package/x11r7?id=8118da144991ab7004522b1e16f06675cc31b3ec) missing. Can you add them please?

lpt2007, 2019-03-18:
@pvizeli. Is it possible to add packages xorg and x11r7 in next release? If not can I add them manually?

dap35, 2019-03-22:
Just my two cents... The only GUI for HA is the web UI.   Hassio is designed to be a headless.   I'm not sure what you are trying to accomplish by running X Windows.

lpt2007, 2019-03-24:
Hi @dap35 

I try to make HA addon to start chrome browser on hassos host with monitor attached to hdmi port.

pvizeli, 2019-04-05:
@dap35 is right. You can try to build an add-on they give output to HDMI as you want.

closed 2019-04-05 by pvizeli

lpt2007, 2019-04-05:
Hi,

I already made hassio addon KODI with output to hdmi. Addon is in early stage but works great.

Here is link if someone want to try: [LINK](https://github.com/lpt2007/its-ai/tree/master/apps-hassos-addons/kodi/old_v0.2a)

I think addon probably work because KODI is not depended on x librarys to work.

When I tried to make addon to start chrome with some page. I get error display not found. (I also try with magicmirror / retropie / firefox and they are all x depended, and I have similar errors about display in all of them).

Then I do some research about docker and I think the problem is that hostOS(buildroot) does not have installed some minimal x librarys to start x display when docker need them.