home-assistant/operating-system #377: Not working with Xenserver

john-arvid, 2019-03-14:
Hi,

Not sure if this is related to #265 and #120 
But I think since xenserver not supports UEFI boot and the hassos image require it, it will not work on xenserver.
Even if #265 refers to implementing Xen.

I have tested and it will not boot past the bios.

pvizeli, 2019-03-18:
My XenServer 7.2 support UEFI?

john-arvid, 2019-03-18:
Yes, but does your vm's? Ref: https://github.com/xcp-ng/xcp/issues/16#issuecomment-429300423

knd775, 2019-03-18:
> My XenServer 7.2 support UEFI?

Without heavily modifying XenServer (and potentially recompiling the kernel), booting VMs in UEFI is not possible.

john-arvid, 2019-03-19:
Maybe we could get a image with legacy boot instead of UEFI?

closed 2019-05-25 by stale[bot]

DaAwesomeP, 2019-06-05:
I wish this was not automatically closed. What needs to be done to make a legacy boot build? xcp-ng is the most popular open source hypervisor.

john-arvid, 2019-06-05:
It looks like UEFI will be supported in version 8. I haven't tested this feature yet.

[https://github.com/xcp-ng/xcp/issues/16#issuecomment-498981998](https://github.com/xcp-ng/xcp/issues/16#issuecomment-498981998)

But it would be nice to have a build without UEFI as you say.

DaAwesomeP, 2019-06-06:
I don't know that version 8 will be released for a while. If you can point me in the right direction, I can add a build for legacy boot. Unfortunately, I don't have any experience with building images like this, so if you know which files I need to look at, I can do some research.

DaAwesomeP, 2019-08-13:
I can confirm that HassOS works with XCP-ng 8.0.0. However, it does not seem possible to install the guest OS tools due to the read-only filesystem.

jstorgaard, 2020-08-10:
Anyone got any luck installing Xen guest tools for XenServer/XCP-NG?
(Ref. https://github.com/home-assistant/operating-system/pull/265)

Brandon4466, 2020-08-18:
@jstorgaard how did you get it too boot?

Whether I use the VHDK or the OVA it just says Boot from "Hard Drive... Success" and just stays there.

TFyre, 2021-04-30:
I had to change from `bios` to `uefi` on XCP-ng 8.1

`xe vm-param-set uuid=VM-UUID HVM-boot-params:firmware=uefi`