home-assistant/operating-system #383: update error

ifly3years, 2019-03-22:
when i update my hassos , 
use the "update" link in system ---->host system, not the
>>>>http://192.168.15.XX:8123/hassio/system

 i got  this info ;
____________________________________________________
Error: 500 Internal Server Error Server got itself in trouble

and system log said:
19-03-22 10:57:52 INFO (MainThread) [hassio.hassos] Fetch OTA update from https://github.com/home-assistant/hassos/releases/download/2.10/hassos_ova-2.10.raucb

______________________________________________________

i think，something was wrong, because ,i have ht 3.0 hassos, not the 2.10;
so, how can i find the code and repair it? 
thanks

ifly3years, 2019-03-22:
oh，no!!!

i click the update button,,,,,and this time it's ok, 
he did it , an downgrade my system to 2.10......

oh, ,,,,,,

pvizeli, 2019-04-05:
Yes, this button is only for stable development circle.

closed 2019-04-05 by pvizeli