home-assistant/operating-system #385: Allow changing the time source in the web ui

rradar, 2019-03-25:
Open this one in relation to the problem that I can't keep time in a offline enviroment and RTC's (#372) are not (yet?) supported.

Would be nice to have a possibility to point my hassio (raspi) installation to another raspi with ntp server and RTC to keep the time. Best would be if it also would support the rtc so I would not need to run a secondary raspi for but for the beginning? :grin:

rradar, 2019-04-02:
And how can I change the time source manually? I found a file (was it something like timesyncd?) in the hassio-overlay with some google domains and added a local ntp server in the beginning.

But guess what? Nothing. Just keeps running with the wrong time...

I also was not able to do any debugging because I couldn't find any log (hassio, homeassistant) contain information about ntp/time syncing.

Any help would be nice! :+1:

By the way: How can I do *full* offline updates of hassio/hassos/hass&addons? For hassos there is a button to get updates from usb but where do I get the update files and what is with the rest of the show? :fireworks:

rradar, 2019-04-06:
I think I need to migrate back from hassio to home assistant running in virtual env :disappointed: 

It's just not possible to get proper time sync without internet (and without google!) in hassio. Said for a software which self claims to "put **local control** and **privacy first**". Both is **not possible** if you once rely on internet and second rely on google! :lock: 

There is no chance to run home assistant proper without have a proper time sync. :clock1030:

gervaso, 2019-05-02:
I have the same module connected to the gpios, but looks there's no way to "interact" with hassos to load the proper kernel module.

rradar, 2019-05-07:
I left the hassos where it is (to much trouble in this walled garden) and went with hassio install based on raspbian. There I have the freedom to use the rtc and set custom ntp server.

gervaso, 2019-05-07:
@rradar that is where I came from, moved to hassos just to get rid of the "infrastructure management". I think it is supported, as stated in the documentation, just not really user friendly. I'll try the "CONFIG" usb device and debug ssh access on port 22222.

rradar, 2019-06-24:
@gervaso did you had any success?

francescofabbro89, 2019-07-01:
@gervaso  how did you manage to do that?
did you load hasso on a docker container? and in this way does the RTC work?

closed 2019-09-06 by stale[bot]

yinyue200, 2020-04-20:
/active 
any news

francescofabbro89, 2020-04-27:
no