home-assistant/operating-system #387: Include NTC Server functionalities with RTC support

rradar, 2019-03-27:
I thought once again how to keep time in my true offline environment.

And the best would be if hassio could not only support RTCs (#372)and would be able to act as configurable NTP client (#385) but also to act as a NTP server.

Because the majority of my sensors (for example all esp's) don't have a RTC on they own and rely on a time source. Best would be if hass act as the universal time source for all. So there will be for sure no miss match in time between server (hassio) and clients (esphome etc.)

pvizeli, 2019-04-05:
We use systemd timesync because the root file system is RO and it needs a system like that from systemd to store times over reboot.

Feature request about the functionality of systemd needs to be inside there repository. We support only what they provide to us.

pvizeli, 2019-04-05:
You can also make a PR and integrate a new time backend they run on a RO file system. I'm open to change the time backend. Otherwise please close the PR.

rradar, 2019-04-07:
@pvizeli thank's for your feedback. But I have really no clue about coding. I'm still not able to change even the NTP server from our big brother google to a local NTP server in my network :sad: Maybe you have an idea where (which files etc.) to look for to for. You could reply here #385 if you have any idea.

I'm going to close this one.

closed 2019-04-07 by rradar