home-assistant/operating-system #388: VMDK: Docker top error

tboyce021, 2019-04-01:
Running `docker top homeassistant` on the host gives the following error:

```
ps: ps: invalid option -- 'e'
```

Using the VMDK with Proxmox on a NUC:
```
 System: HassOS 2.11  (amd64 / qemux86-64)
 Home Assistant version: 0.90.2
 Supervisor version: 152
```

This seems to be preventing the Portainer add-on from displaying container stats (see [hassio-addons/addon-portainer#3](https://github.com/hassio-addons/addon-portainer/issues/3#issuecomment-478739830)).

pvizeli, 2019-04-05:
You can use our API: /homeassistant/stats

closed 2019-04-05 by pvizeli

tboyce021, 2019-04-29:
I'm not sure the API is sufficient for this. The Portainer addon allows for installing containers that are **not** addons, but the API won't return the stats for those. Also, I'm not sure if there's a way to customize Portainer to use that API instead of docker top/ps.