home-assistant/operating-system #392: bug report Snapshot

RWGaMeR, 2019-04-11:
Good morning,
I want to report a very annoying bug, when I restore the snapshot it does not load external repositories and therefore does not load third-party addons, can you fix this annoying bug?

thank you,
Francesco Russo

______________________________________________________________________________________________________

my web site: https://www.itchsblog.it

papperone, 2019-04-12:
+1
I've noticed the exact same problems when trying to move my hassio to a new hardware; I crate a snapshot, reloaded the snapshot on a fresh new installation and all is good but only for the add-ons which are part of the Official Repository; all add-ons which have been installed from other repositories must be manually added (prior to install all the missing repositories) and then manually configured or reloding again the snapshot.

I think this is a bug missing feature (if you don't want to call it "bug") as I'm expecting that as the snapshot includes all add-ons and their configuration woudl carry as well a backup of all the additional reporitories configured in Hass.IO; this will make restoring a snapshot an easier process.

RWGaMeR, 2019-04-12:
@homeassistant

pvizeli, 2019-04-12:
The repository will only restore on full restore and not partial. That is no bug.

That is also the wrong repository, please open an issue on Hass.io repository of supervisor. I put it to my task list for Q3-Q4 2019 if no other developer tries to improve the partial restore.

closed 2019-04-12 by pvizeli

RWGaMeR, 2019-04-12:
@pvizeli 
ok, then I'll move the question

papperone, 2019-04-12:
> 
> 
> The repository will only restore on full restore and not partial. That is no bug.
> 
> That is also the wrong repository, please open an issue on Hass.io repository of supervisor. I put it to my task list for Q3-Q4 2019 if no other developer tries to improve the partial restore.

maybe I did not explain myself correctly but the issue I experience was on full snapshot recovery and definitely it's a bug...
the process did not work as the addons not in the offcial  repository weren't restored!