home-assistant/operating-system #393: Network becomes disconnected randomly

forast3r, 2019-04-21:
Running hassos on Virtualbox, network becomes disconnected randomly (last time the machine has been up for exactly 24 hours.
Last time it happened, I upgraded from HassOS 2.9 to HassOS 2.11. 
Today I forced a restart of the network adapter (with a double toggle of Virtualbox "Cable Connected" option). The network is configured as Bridged, IP address is assigned in the router with DHCP.

After checking the functions, HA can connect to all the devices in the LAN, but I can't see no logs in Hass.io/System Log.
The Supervisor can't be restarted from web interface. 

Addons aren't functional too (I guess the connection with docker internal networks is down)
Accessing the VM console I am able to login and check dmesg, with some messages about an IPV6 interface (sorry, can't copy text from remote desktop)

Routing table (ip r) appears to be empty

Are there some other logs to check?

andrewreid, 2019-06-27:
Did you get anywhere with this? I've had a similar issue that I thought I'd resolved, but it's cropped up again since the latest update.

Some months ago when I first spun up my Hass.io instance in a bhyve VM, I was experiencing this random network drop-out issue you're describing. It _felt_ like a link-level issue, and a bit of reading at the time led me to try changing the virtual NIC from an `e1000` to `VirtIO`. That resolved it completely and it was smooth sailing until a recent upgrade.

I'm now noticing the same random drop-outs occurring again, even though I'm using the `VirtIO` NIC, and it does the same thing with `e1000`. Like you, I wasn't able to grab `dmesg` output from my VNC console.

I guess the question is, is this a hypervisor or HassOS issue?

forast3r, 2019-06-28:
I couldn't try much more than you. Given the zero feedback, I finally I migrated this setup to a debian+docker ce virtual machine. No hiccups since then.

pvizeli, 2019-06-28:
I don't known. We update the NetworkManager on 3tree and Kernel. Maybe it fixit

closed 2019-09-03 by stale[bot]