home-assistant/operating-system #407: Snapshot not restoring

ndarilek, 2019-05-14:
Made a snapshot under 2.11, tried restoring under 3.1. Only my HA release, configuration, and a single addon restored. No extra repositories or addons were configured.

Pretty sure I reported something similar a while ago and it was declared fixed. How do we debug this? Of all the functionality to fail, restoring a snapshot isn't on the list of things that should go wrong.

pvizeli, 2019-05-14:
`su logs` on console

cogneato, 2019-05-20:
You do need to add third party repos to the addon store before restoring any such addons.

ndarilek, 2019-05-20:
Why, aren't they part of the backup?

I'll get some logs later today or tomorrow, been swamped at work and 
hadn't counted on needing to rebuild my server.

ndarilek, 2019-05-20:
Here are logs of a snapshot restore. I've confirmed that the repositories for third-party addons are listed in the snapshot.json. These should presumably be added before the Docker images are retrieved.

```
[32m19-05-20 22:16:47 INFO (MainThread) [__main__] Initialize Hass.io setup[0m
[32m19-05-20 22:16:48 INFO (MainThread) [__main__] Setup HassIO[0m
[32m19-05-20 22:16:48 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to Supervisor homeassistant/armv7-hassio-supervisor with version 164[0m
[32m19-05-20 22:16:48 INFO (MainThread) [hassio.utils.gdbus] Introspect org.freedesktop.systemd1 on /org/freedesktop/systemd1[0m
[32m19-05-20 22:16:48 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd1[0m
[32m19-05-20 22:16:48 INFO (MainThread) [hassio.utils.gdbus] Introspect org.freedesktop.hostname1 on /org/freedesktop/hostname1[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.hostname1 - /org/freedesktop/hostname1[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Introspect de.pengutronix.rauc on /[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: de.pengutronix.rauc - /[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.host.info] Update local host information[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.DBus.Properties.GetAll on /org/freedesktop/hostname1[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.host.services] Update service information[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ListUnits on /org/freedesktop/systemd1[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.host.services] Reload local service hassos-apparmor.service[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.utils.gdbus] Call org.freedesktop.systemd1.Manager.ReloadOrRestartUnit on /org/freedesktop/systemd1[0m
[32m19-05-20 22:16:49 INFO (SyncWorker_1) [hassio.docker.interface] Attach to homeassistant/raspberrypi3-homeassistant with version 0.92.2[0m
[32m19-05-20 22:16:49 INFO (MainThread) [hassio.hassos] Detect HassOS 3.1 on host system[0m
[32m19-05-20 22:16:49 INFO (SyncWorker_0) [hassio.docker.hassos_cli] Found HassOS CLI homeassistant/armv7-hassio-cli with version 9[0m
[33m19-05-20 22:16:50 WARNING (MainThread) [hassio.store.data] Can't read /data/addons/local/mopidy/config.json: required key not provided @ data['arch']. Got None[0m
[32m19-05-20 22:16:55 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/core repository[0m
[32m19-05-20 22:16:55 INFO (MainThread) [hassio.store.git] Load add-on /data/addons/git/a0d7b954 repository[0m
[33m19-05-20 22:16:55 WARNING (MainThread) [hassio.store.data] Can't read /data/addons/local/mopidy/config.json: required key not provided @ data['arch']. Got None[0m
[32m19-05-20 22:16:55 INFO (MainThread) [hassio.store] Load add-ons from store: 59 all - 59 new - 0 remove[0m
[32m19-05-20 22:16:55 INFO (MainThread) [hassio.addons] Found 8 installed add-ons[0m
[32m19-05-20 22:16:56 INFO (SyncWorker_5) [hassio.docker.interface] Attach to bestlibre/armv7-syncthing with version 1.1.0[0m
[32m19-05-20 22:16:56 INFO (SyncWorker_1) [hassio.docker.interface] Attach to hassioaddons/appdaemon3-armv7 with version v3.0.1[0m
[32m19-05-20 22:16:56 INFO (SyncWorker_3) [hassio.docker.interface] Attach to homeassistant/armv7-addon-ssh with version 5.4[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.updater] Fetch update data from https://s3.amazonaws.com/hassio-version/stable.json[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.snapshots] Found 1 snapshot files[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.discovery] Load 0 messages[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.ingress] Load 0 ingress session[0m
[32m19-05-20 22:16:56 INFO (MainThread) [__main__] Run Hass.io[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.misc.dns] Start DNS port forwarding for host add-ons[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.api] Start API on 172.30.32.2[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.addons] Phase 'initialize' start 0 add-ons[0m
[32m19-05-20 22:16:56 INFO (MainThread) [hassio.addons] Phase 'system' start 2 add-ons[0m
[32m19-05-20 22:16:57 INFO (MainThread) [hassio.host.alsa] Update ALSA device list[0m
[32m19-05-20 22:16:57 INFO (MainThread) [hassio.host.alsa] Detect input device 0,0[0m
[31m19-05-20 22:16:57 ERROR (SyncWorker_4) [hassio.docker] Can't create container from addon_02defeca_pulseaudio: 404 Client Error: Not Found ("No such image: nolan/hassio-pulseaudio-armhf:latest")[0m
[32m19-05-20 22:16:57 INFO (SyncWorker_13) [hassio.docker.interface] Clean addon_53caca22_syncthing application[0m
[32m19-05-20 22:17:05 INFO (SyncWorker_13) [hassio.docker.addon] Start Docker add-on bestlibre/armv7-syncthing with version 1.1.0[0m
[31m19-05-20 22:17:05 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished coro=<Addon.start() done, defined at /usr/src/hassio/hassio/addons/addon.py:459> exception=AddonsError()>[0m
Traceback (most recent call last):
  File "/usr/src/hassio/hassio/addons/addon.py", line 479, in start
    raise AddonsError() from None
hassio.exceptions.AddonsError[0m
[... cut ...]
[32m19-05-20 22:21:23 INFO (MainThread) [hassio.api.proxy] Home Assistant WebSocket API request initialize[0m
[32m19-05-20 22:21:23 INFO (MainThread) [hassio.api.proxy] WebSocket access from a0d7b954_appdaemon3[0m
[31m19-05-20 22:21:23 ERROR (MainThread) [hassio.api.proxy] Client error on WebSocket API Cannot connect to host 172.30.32.1:8123 ssl:False [Connection refused].[0m
[32m19-05-20 22:21:25 INFO (MainThread) [hassio.api.security] /supervisor/logs access from core_ssh[0m
```

Thanks.

ndarilek, 2019-05-22:
As I restore this backup again and again, more addons seem to appear in 
my dashboard. But they can't be started, so I suspect their images 
aren't being pulled. It's odd that they'd appear if that is the case.

I'll check the images on my host tomorrow and see if they're in fact 
getting pulled.

ndarilek, 2019-07-21:
Please don't close, as this is definitely still a thing, and this is the 
second time I've opened an identical issue. I can once again provide a 
backup on request--I did so the first time but never heard anything in 
response. Not sure what more I can do to help debug this, and it worries 
me that an issue as serious as snapshots not restoring isn't taken more 
seriously.

ndarilek, 2019-09-19:
Still important IMO. Backups...

My offer of a failing backup still stands. Because of the nature of a 
backup, I'd rather not provide it to just anyone. I know I've offered 
one in the past but never heard back about it.

ndarilek, 2019-11-18:
*boing*

kajmaj, 2019-12-11:
3.5 does not read snapshots from 2.12 at all.
Is there any way around?

ndarilek, 2020-02-10:
...