home-assistant/operating-system #411: Odriod C2 2.12 or 3.1 img cannot get network link up

namiba, 2019-06-19:
I downloaded the latest 2.12 as well as 3.1 of Hassos for the C2. Flashed SD card with etcher and viewed the boot messages via console, but there is no network link on C2 or switch with the default DHCP config. I also tried with a static IPv4 but still no network link.

On the console debug shows a lot of IPv6 and link not ready and finally shows link ready but the link light on the C2 or switch does not light. I cannot ping out after using nmcli and setting the C2 to static IP. 

Has anyone tried with these versions on the Odriod C2?

rbray89, 2019-06-20:
hmmm... I'm running on 2.11 without any problems. Can you try that version quickly to eliminate config or HW issues?

rbray89, 2019-06-20:
Hmmm... Just performed an OTA update to 2.12, and everything looks good here. Did you create a network config for it? Syntax errors can cause this problem. Try deleting any network configs already present and reboot to restore defaults if you haven't already.

closed 2019-08-26 by stale[bot]