home-assistant/operating-system #412: Raspberry Pi 2 Model B does not boot

BenLiyanage, 2019-06-23:
Hello,

When flashing my sd card with the following images, the raspbery pi does not boot.  No signal is sent to the HDMI cable, and there is no link light activity on the Pi or my switch.

I have confirmed that my hardware is legitimate by using Raspbian Stretch Lite (https://www.raspberrypi.org/downloads/raspbian/) and seeing it boot.

I tried the following images: 

* https://github.com/home-assistant/hassos/releases/download/2.12/hassos_rpi2-2.12.img.gz
* https://github.com/home-assistant/hassos/releases/download/2.11/hassos_rpi2-2.11.img.gz

Pi Version Information:
```
pi@raspberrypi:~ $ grep Revision /proc/cpuinfo
Revision        : 000e
```

https://elinux.org/RPi_HardwareHistory

    000e     Q4 2012     B     2.0     512 MB     (Mfg by Sony)

I am fairly technical and can likely make the PR to fix this if someone can point me towards troubleshooting techniques for raspberry pi images not booting.

lmalsch, 2019-08-24:
@BenLiyanage did you found a solution?
I have the same problem with my Raspberry Pi 2 Model B.

BenLiyanage, 2019-08-30:
My solution was to install the python packages manually. Not really ideal.

closed 2019-11-05 by stale[bot]