home-assistant/operating-system #413: Raspberry Pi 4

pepeEL, 2019-06-24:
Hi
Now is released Raspberry Pi 4 :) 
When we can think that image for Raspberry Pi 4 will be available ?

homeassistantmi2, 2019-06-29:
Can yo release immediately hassos on Rpi4 ? 
Only option now is use manual installation on linux
ResinOS it also not work on rpi4

Joeboyc2, 2019-06-29:
Just wait a little bit, give the devs a chance!

pepeEL, 2019-06-29:
And good will be add support boot from ssd on usb3. 0
This is very need.

pvizeli, 2019-07-01:
We can first add support for RPi4 if u-boot (our bootloader they we use because we are a embedded system) have a support patch.

skynet01, 2019-07-03:
Would be amazing if we can just take out our sd  card from Rpi3 and stick it in Rpi4 and it just starts working like nothing happend 🥰

Joeboyc2, 2019-07-03:
I think that's an impossibility due to the changes thr new hardware has.

Nothing wrong with starting again 😁

pvizeli, 2019-07-16:
@Joeboyc2 have right, there are complete different bootloader and microcode to setup the CPU and start the kernel. Raspbian work a bit different, they can handle that because they but everthing into same image and support only 32bit.

pvizeli, 2019-07-16:
Support start with 3.3

closed 2019-07-16 by pvizeli

skynet01, 2019-07-16:
Woohoo! Thank you!

Joeboyc2, 2019-07-16:
I'll be flashing it tonight, will let you know of I have any issues 😁

Joeboyc2, 2019-07-16:
oh.... @pvizeli i dont see the pi 4 image.... which one should i use to flash ? the pi3 one?

MrYadro, 2019-07-16:
You just have to wait a bit while they are building, now they are up in assets

skynet01, 2019-07-16:
So stoked to try it. Is there going to be a migration tutorial of some sort from Pi3+? Or you just flash it and restore a snapshot?

Joeboyc2, 2019-07-16:
Ah i see them now, literally only 22 assets when i check and sent the message, 24 now :)