home-assistant/operating-system #422: Intel NUC won't boot off NVME M.2 drive

bruhautomation, 2019-07-01:
Hello! Thanks for the awesome work on HassOS. 

I'm trying to run HassOS off of a NUC. Everything works great when I use a standard SSD. I burned the image directly to the drive using balenaEtcher via a USB-to-SATA adapter. Pop the drive into the NUC and it's rolling. 

However, when I repeat this exact process with a NVME M.2 drive, HassOS refuses to boot. I've tried copying the partitions from the working drive to the NVME drive, but that didn't resolve the issue. 

I have the secure boot settings turned off in the BIOS and confirmed the NUC can see the NVME M.2 drive. I've scrubed the drive and confirmed that all the partions are being written identically to the working standard SSD. 

No matter what I try, HassOS on the NVME M.2 drive tries to boot, but ultimately leaves me stuck at HassOS boot menu. 

Do you have a guess why the switch the NVME M.2 drive prevents HassOS from booting/running? 

I'm wondering if the partition labels are affecting something. 

The working SSD uses /dev/sda1 /dev/sda2, etc. The not-working NVME drive uses /dev/nvme0n1p1 /dev/nvmeon1p2 etc. 

Is HassOS looking for the standard partition labels on boot? 

Thanks much!

botne, 2019-08-21:
Hi!

I have the same issue.
Did you find a workaround for this?

Br,
Frank

botne, 2019-08-26:
ok, @bruhautomation. Will change the m.2 nvme for a m.2 sas instead. I guess that will fly :)

pvizeli, 2019-08-28:
I think it's a barebox issue. I hope uboot come also soon with an working UEFI support

rovrell, 2019-09-27:
Ahh, that's the problem :/ tried for 2 days now.

xbmcnut, 2019-11-14:
Hey Ben. Happening for me too. Sorry, I duplicated the issue here https://github.com/home-assistant/hassio/issues/1365

SBerg1980, 2019-11-27:
I have the same problem, i want to switch from my pi to my NUC with a nvme disk but when i run de gunzip and then boot the NUC its stuck on boot.

air720boarder, 2019-12-22:
This is currently a blocker for me as well

Mathijs0, 2019-12-23:
This is why I now use Debian with hass.io and not hassos

air720boarder, 2019-12-24:
Any fixes in the pipeline for this?

Mathijs0, 2019-12-24:
Ore a warning on the install page?

niXta1, 2019-12-26:
Also got stuck here...

AdamNaj, 2020-02-05:
Same problem for me... Was so excited when the NUC arrived, didn't last long...

niXta1, 2020-02-06:
> Same problem for me... Was so excited when the NUC arrived, didn't last long...

Install a sata ssd or go with another OS... Only options for now...

manavees, 2020-02-07:
Any news about how to run hass in nvme ssd on a NUC? I am thinking of getting a NUC for my home assistant and would like to go with nvme

xbmcnut, 2020-02-07:
You'll have to run something like Promox as I ended up doing. Check out Drzzs videos in that process.

manavees, 2020-02-07:
> You'll have to run something like Promox as I ended up doing. Check out Drzzs videos in that process.

What about using a 2,5'' SSD? Will it work without any issues?

xbmcnut, 2020-02-07:
Should do no issue although some reported that 3.8 was causing issues on my YouTube channel. Maybe 3.9 resolved this?

manavees, 2020-02-07:
So basically also a simple m.2 sata could work. Like the Samsung (https://s3.ap-northeast-2.amazonaws.com/global.semi.static/Samsung_SSD_860_EVO_M2_Data_Sheet_Rev1.pdf)

niXta1, 2020-02-07:
> So basically also a simple m.2 sata could work. Like the Samsung (https://s3.ap-northeast-2.amazonaws.com/global.semi.static/Samsung_SSD_860_EVO_M2_Data_Sheet_Rev1.pdf)
No, that has a m2 interface, it needs to have a SATA connector (2.5").

Mathijs0, 2020-03-31:
Is there any progress in this issue?

Blankf, 2020-04-08:
Same question here, is there any progress on this issue? Or is someone looking at it?

tradiuz, 2020-04-08:
When you burn the NVMe with etcher, what do you see under `/dev` in the HassOS boot menu.  I was able to get my AtomicPi to boot manually by mounting `/dev/usbdisk1.hassos-kernel0` to `/mnt/system` because the system0 script assumes the drive is named `disk*.hassos-kernel0`  I suspect the NVMe drives show up similarly different from a "disk".

A stupid fix might just be changing the `disk*` to just `*` or some regex of all possible drive types (mmcblk, nvmeon, disk, usbdisk, etc).

Blankf, 2020-04-09:
How were you able to mount it from the shell?

i see a bunch of 
disks0/1/2/3/4/5/6/7 and
disk1/2/3/4/5/6/7
disk0.hassos-boot
disk0.hassos-bootstate
disk0.hassos-data
disk0.hassos-kernel0
disk0.hassos-kernel1
disk0.hassos-overlay
disk0.hassos-system0 en 1

not seeing anything that looks like nvme.

closed 2020-04-18 by pvizeli

Mathijs0, 2020-04-21:
Nice and thank you! I am gone try it out soon.

Mathijs0, 2020-04-25:
O this issue is closed but not jet released, when wil this fix be implemented? Now that I formatted my drive :-(

frenck, 2020-04-25:
Yes, issues close when development is finished and does not tell anything about availability.

We don't have a schedule for this. Things are ready when they are ready.

Mathijs0, 2020-04-25:
okey thanks for your work!!

Joc007, 2020-05-02:
Success, it works now.
If you need details, see this: https://community.home-assistant.io/t/my-steps-to-success-for-hass-io-on-intel-nuc/43915/323

Infraviored, 2023-02-13:
I have the same issue. Still, i do not understand how to apply this community post 'guide' - it lack so much information.
Instead of using etcher, use the restore disk feature built into ubuntu?

PR #626 (2020-04-18): Support NVMe & update barebox
fix: #422