home-assistant/operating-system #434: No Add-ons allowed on Raspberry Pi 4

robpow, 2019-07-17:
Using hassos_rpi4-3.3.img.gz from 20190716.2 on a Raspberry Pi 4 and joined the dev channel. All Add-ons, both official and community) are unavailable. 

The log output if you try to install any add-on is like this:
`19-07-17 12:39:12 ERROR (MainThread) [hassio.addons] Add-on a0d7b954_logviewer not supported on that platform`

System info:
`
arch | armv7l
-- | --
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.4
timezone | Europe/Stockholm
version | 0.96.0b4
virtualenv | false
`

robpow, 2019-07-17:
Not sure if it's an issue with how the architecture is tagged or what architectures all the add-ons are allowed on?

lenisko, 2019-07-17:
Which is interesting, as `arch` is reporting working architecture.
```
# /bin/arch
aarch64
```
might be some miss-configuration

robpow, 2019-07-17:
On a 32-bit RPi 4 though it returns:
`# /bin/arch`
`armv7l`

robpow, 2019-07-17:
Same error with the 64-bit build:

`19-07-17 19:27:02 ERROR (MainThread) [hassio.addons] Add-on a0d7b954_traccar not supported on that platform`

Setting | Value
-- | --
arch | aarch64
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.4
timezone | Europe/Stockholm
version | 0.96.0b4
virtualenv | false

robpow, 2019-07-17:
Could this PR be related?
[https://github.com/home-assistant/hassio/commit/5e8007453f98f57fa2b1f948711235edfe243b91#diff-14aba126950bc23275de21fa6270e0d4](url)

zbuh, 2019-07-18:
Installed manually on top of raspbian, and got the same issue.

jamiepryer, 2019-07-18:
same here.
rpi4, 4gb, 64bit HassIO

arch | aarch64
-- | --
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.4
timezone | Europe/London
version | 0.96.0b3
virtualenv | false

jamiepryer, 2019-07-18:
Same goes for 32bit too

robpow, 2019-07-19:
Confirmed fixed!
The upgrade from 169 to 170 however left home assistant stopped and I had to go into the hassio client to start it.