home-assistant/operating-system #435: Broken WiFi on Raspberry Pi 4 / HassOS 3.3 64bit

lenisko, 2019-07-17:
Like in title, looks like WiFi is not working for latest (beta) version of HassOS for rpi4.
```
hassio >
arch: aarch64
channel: stable
hassos: "3.3"
homeassistant: 0.96.0b3
hostname: hassio
logging: info
machine: raspberrypi4-64
supervisor: "169"
supported_arch:
- aarch64
- armv7
- armhf
timezone: UTC
# uname -a
Linux hassio 4.19.58-v8 #1 SMP PREEMPT Tue Jul 16 21:21:05 UTC 2019 aarch64 Hassio/OS
# nmcli
eth0: connected to Wired connection 1
        "eth0"
        ethernet (bcmgenet), HIDDEN, hw, mtu 1500
        ip4 default
        inet4 192.168.1.238/24
        route4 0.0.0.0/0
        route4 192.168.1.0/24
        inet6 fe80::ae04:f1e9:a4b1:aeb7/64
        route6 fe80::/64
        route6 ff00::/8

docker0: unmanaged
        "docker0"
        bridge, 02:42:AF:E0:28:B0, sw, mtu 1500

hassio: unmanaged
        "hassio"
        bridge, 02:42:55:47:C1:D2, sw, mtu 1500

vethd122498: unmanaged
        "vethd122498"
        ethernet (veth), 6E:3A:AA:2F:08:E9, sw, mtu 1500

vethe815468: unmanaged
        "vethe815468"
        ethernet (veth), 76:5D:85:91:4B:67, sw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

DNS configuration:
        servers: 192.168.1.1
        domains: lan
        interface: eth0

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(5) manual pages for complete usage details.
# nmcli radio
WIFI-HW  WIFI     WWAN-HW  WWAN
enabled  enabled  enabled  enabled
# nmcli device wifi rescan
Error: No Wi-Fi device found.
# nmcli device wifi
# dmesg  | grep brcmfmac
[    1.930670] brcmfmac: F1 signature read @0x18000000=0x15264345
[    1.935632] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    1.936832] usbcore: registered new interface driver brcmfmac
[    3.043709] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    4.054813] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
#
```

pvizeli, 2019-07-17:
Yeah, I wait for new drivers there

Quentintin, 2019-07-24:
Do you think it will come soon ?

I didn't succeed to make bluetooth works too. Does it work with beta version ?

pvizeli, 2019-07-25:
I had no issue with bluetooth

pcopa, 2019-08-01:
Just installed my first pi4 with 1G and eth0 works fine, but can't get wlan0 to work. not with CONFIG/network/my-network file on boot parition OR with CONFIG/network/my-network on a USB sick (the same USB that works for pi3). Unfortunately can't give more info, need a display to HDMI cable.

lenisko, 2019-08-01:
@pcopa https://github.com/home-assistant/hassos/issues/435#issuecomment-512548472

haruny, 2019-08-03:
I wish I read this before spending 2 hours on making wifi work.

Tomas0413, 2019-08-06:
I'm having the same issue with 32bit version. I opened a new topic on the forum site a few days ago and just found this issue on GitHub.
https://community.home-assistant.io/t/no-wlan0-interface-on-raspberry-pi-4-model-b-32bit/129984/7

doudz, 2019-08-09:
I confirm the issue with 32bit version too on Pi4 2G, any update ?

libo, 2019-08-10:
@pvizeli purely for educational purpose and not to put any pressure, where are we waiting the drivers from? I am trying to better understand the architecture, do we use the same kernel and firmware as raspbian? If so could we use their driver?

pedrobsm, 2019-08-19:
Same issue here with a Pi4 2GB. It seems the wi-fi driver is still not working. Solved it by pluging in an old Edimax USB Wi-fi adapter took from my Pi 1 Model B. :)

vencakratky, 2019-08-22:
I have the same Problem. After 3 Hours i found this forum . Bad Day

closed 2019-08-24 by pvizeli

eftibug, 2019-08-26:
Hello,
Just realized why I was pulling hairs for hours!!!
Do you know when we would be able to download a new image for HassOS?
Thanks!

cigas4, 2019-08-29:
Same issue on Raspberry Pi 4 model B 4Gb, Hass.io 3.3 64bit.
But Wifi works fine on same pi4, using Hassbian v1.6.1 image (2019-07-02 release).
Looking forward to have an update for Hassio!

TheZoker, 2019-08-29:
@cigas4 3.4 which includes the fix was just released

cigas4, 2019-08-29:
> 
> 
> @cigas4 3.4 which includes the fix was just released

I have just updated to beta 3.4 release and can confirm that wifi is now working on Raspberry Pi 4 model B 4Gb, Hass.io 3.4 64bit.

cigas4, 2019-08-29:
Is there a way to setup multiple wifi SSIDs?
If I take my pi other location I need to connect to other wifi. Can I setup few SSIDs in my-network file so that hassio would connect to one which is available from the list?
Or every time I need to play with USB stick uploading new my-network file with changed SSID and pass?

iBobik, 2019-08-29:
@cigas4 It should be possible to have multiple files like that called my-network - the name is just example. Off topic here, btw. :-)

Tomas0413, 2019-08-29:
The fix works for me too on Raspberry Pi 4:
kernel: 4.19.67-v7l
operating_system: HassOS 3.4

vencakratky, 2019-08-29:
Can you please advise me, how to install update to 3.4  in home assistant via SSH & Web terminal?
I tried this link, but it writes me an error
"pip3 install --upgrade git+https://github.com/home-assistant/hassos/releases/download/3.4/hassos_rpi4-3.4.img.gz"
after clone but with error "cannot detect archive format
Thank you all

cigas4, 2019-08-29:
> Can you please advise me, how to install update to 3.4 in home assistant via SSH & Web terminal?
> I tried this link, but it writes me an error
> "pip3 install --upgrade [...]

Download image file .raucb format
Copy image file to root of USB stick (FAT32 format)
Navigate to HA UI HASS.IO -> System -> Inport from USB
Reboot

pvizeli, 2019-08-29:
`hassio os update --version 3.4`

vencakratky, 2019-08-29:
> 
> 
> `hassio os update --version 3.4`

WOW  After update and reboot, wifi connected with "nmcli device wifi connect ..."
Thank you! 👍

cigas4, 2019-09-10:
Hey guys,

Any of you have an issue with wifi connection loss on Raspberry Pi 4, Hass.io 3.4?
Wifi connection itself works fine, I can connect to my home network and access UI over laptop and phone. Everything works perfect, but after some time (few hours or half of a day) connection stops. Then I cant connect to UI from any device. All sensors works fine, but I cant access UI. So it looks like it looses connection to wifi network. Then only physical reboot of pi helps. After full reboot I can connect again from all devices.
I need to try hooking up pi to Ethernet cable and check if same issue appears.

iBobik, 2019-09-10:
I have no disconnect issue now. I do not know if it is stable, it could restart sometimes without me knowing it, I've just did not found it in state with disconnected WiFi.

Running 3.4 from day after it's release.

Jan Pobořil
https://honza.poboril.cz/

vencakratky, 2019-09-10:
I had only Problem with starting services (MQTT, SSH, Zigbee2MQTT) after boot - If was Raspberry only per Wifi connected. I had to manual start service after each booting. But maybe it was starting Problem with bad SD Card. After new install release 3.4 to new SD Card is Wifi without Problem. Now, what bothers me,i cannnot  find a way, how to boot  Home Assistant on raspberry 4 via USB Disk. But this is other Issue.

mjohngreene, 2019-11-01:
@cigas4 Can you please explain what you mean by "download image .raucb format?"  I can not find anything online about .raucb files.

cigas4, 2019-11-03:
Hi,

See attached photo. There are few versions of each release, img or raucb format.
[Image]
BR,
Andrius

PR #457 (2019-08-24): Rpi update
Fix: #435 #451 #447