home-assistant/operating-system #437: USB devices make serious problems on RPi 3/3B

dreimer1986, 2019-07-19:
I cannot help with much debug logs, but it's easy to reproduce. I tried to update my 2.12 based Hass.IO installation to 3.3 yesterday and itresulted in a mess. First I had serious problems with getting console output after the first (bootloader?) screen. After this was done, all got black forever. Fix was to unplug my CC2530 which seems to cause a fatal hang on bootloader. But that's not the only USB device causing havok. USB Keyboard/Mouse is not detected at all in the whole process causing me not being able to hop into the HASSOS main OS. USB Sticks for a last hope by integrating a public key were not found either. Then when you boot without anything USB like after starting eudev-3.2.8 it loops with lines regarding IPv6: wlan0: link not ready (seems to be known #435) and brcmfmac: power management disabled. More I can not give you, because I had to fix/recreate it. This thing controls too much to stand idle. ^^

rbray89, 2019-07-22:
Might be related to recent FW updates. Have not been able to replicate this on the dev branch (future 3.4 release)

pvizeli, 2019-07-22:
3.4 and 3.3 are the same. I can also not seen an issue on my RPi3

closed 2019-09-27 by stale[bot]