home-assistant/operating-system #439: Rasberry PI 4 only showing 1gb memory

thoj, 2019-07-21:
Hi

I may have misunderstood something and i know the rpi4 image is alpha/beta. But here goes. 

I have a 2gb rpi4. But both /proc/meminfo and top only shows total 923mb memory. I checked with noobs and it shows almost 2gb there. 

dmesg:
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.58-v7l (builder@92b316513952) (gcc version 7.4
.0 (Buildroot 2019.02.3-ge5734a90)) #1 SMP Tue Jul 16 20:31:47 UTC 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 64 MiB at 0x0000000037400000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 46080 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e4 with
 crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41036 r8192 d24500 u73728
[    0.000000] pcpu-alloc: s41036 r8192 d24500 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240960
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1
 security=apparmor rootwait cgroup_enable=memory coherent_pool=1M 8250.nr_uarts=
0 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95x
x.macaddr=DC:A6:32:01:F5:39 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x4000000
0  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff68
59e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 877452K/970752K available (10240K kernel code, 722K rwdat
a, 2580K rodata, 2048K init, 863K bss, 27764K reserved, 65536K cma-reserved, 118
784K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (12256 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 723 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 864 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31797 entries in 94 pages
`

meminfo:
`MemTotal:         945036 kB
MemFree:           27316 kB
MemAvailable:     288636 kB
Buffers:           33496 kB
Cached:           182256 kB
SwapCached:         1312 kB
Active:           367512 kB
Inactive:         329552 kB
Active(anon):     228620 kB
Inactive(anon):   245916 kB
Active(file):     138892 kB
Inactive(file):    83636 kB
Unevictable:           0 kB
Mlocked:               0 kB
HighTotal:        184320 kB
HighFree:            260 kB
LowTotal:         760716 kB
LowFree:           27056 kB
SwapTotal:        236256 kB
SwapFree:          71404 kB
Dirty:               448 kB
Writeback:             0 kB
AnonPages:        480364 kB
Mapped:           131848 kB
Shmem:               104 kB
Slab:             123332 kB
SReclaimable:      54232 kB
SUnreclaim:        69100 kB
KernelStack:        5248 kB
PageTables:         8504 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      708772 kB
Committed_AS:    3548144 kB
VmallocTotal:     245760 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:             1296 kB
CmaTotal:          65536 kB
CmaFree:               0 kB
`

thoj, 2019-07-21:
I guess it has something to do with vc_mem.mem_size=0x40000000

thoj, 2019-07-21:
Just to make sure i tried with the 64bit image. No change. 
I also tried updating the firmware files (bcm2711-rpi-4-b.dtb, start4.elf and fixup4.bin). Also no change.

The vc_mem.* cmdline options comes from start4.elf and is related to how much memory VideoCore gets AFAIK. But I can't for the life of me find any way to override it. I was pretty sure it had something to do with gpu_mem in config.txt but changing that only seems to make even less system memory available.

pvizeli, 2019-07-22:
Yeah, it's strange because we use the same kernel and firmware as raspbian... Maybe there is a systemd script they change the available memory after boot

rbray89, 2019-07-22:
@pvizeli That would be my guess. That or a kernel param. Given the memory varrieties (1, 2, 4GB) They have to do something to detect what is being run and update the available memory sometime at boot.

thoj, 2019-07-22:
Could it have something to do with dtb overlays? 

https://github.com/raspberrypi/linux/blob/618fb8f1d333fbde5695a8086ca9db15d0e72af2/arch/arm/boot/dts/bcm2711-rpi-4-b.dts#L11

https://android.googlesource.com/kernel/msm/+/android-msm-hammerhead-3.4-kk-r1/Documentation/devicetree/bindings/memory.txt

pvizeli, 2019-07-22:
Since we use the device tree from first stage boot loader aka firmware and also they kernel parameters, we handle the startup exact like raspbian. I think it should be deeper on init system maybe?

pvizeli, 2019-07-22:
I found this: https://github.com/raspberrypi/linux/issues/3032

OldUncleMike, 2019-08-07:
Anyone found a fix for this? I also have only 932MB out of 2GB and MemSwap is running at 100%. Would be great to use all the RAM.
The RPi 4 is otherwise massively better/faster than the RPi 3B+

skynet01, 2019-08-14:
Perhaps we can post this issue in the main Raspberry Pi dev? Shouldn't it be something simple as a setting?

pvizeli, 2019-08-14:
Will try later this week an firmware update. Maybe it's fix now

pepeEL, 2019-08-14:
Where is available new FW ?

pvizeli, 2019-08-14:
In buildroot-patch folder

pepeEL, 2019-08-14:
Please give me link....

pvizeli, 2019-08-14:
https://github.com/home-assistant/hassos/tree/dev/buildroot-patches

pepeEL, 2019-08-14:
But option to download file and upload it ?

pepeEL, 2019-08-20:
I check and also on RPI4 with 4GB RAM in Hassio show me only 1GB.

eucciferri, 2019-08-21:
Hi guys!
I just bought an RP4 with 4 GB and I notice that the same thing happens to me (64-bit version of hass.io).
Did anyone manage to solve the problem?
Thank you!

skynet01, 2019-08-24:
Woohoo looks like it got fixed in the latest pull request. When do you think it will be pushed live :)

pvizeli, 2019-08-28:
Fix with: https://github.com/home-assistant/hassos/pull/459

closed 2019-08-28 by pvizeli