home-assistant/operating-system #441: HassOS 2.11: EXT4-FS error

friendodevil, 2019-07-22:
Recently (with 0.95.3, 4 and now with 0.96.x) started getting file system errors: 
EXT4-FS error ext4_mb_generate_buddy:747.

Running HassOS 2.11 in Oracle VM Virtual box. Not sure how to get started with fixing these. Help appreciated.

[image]

pvizeli, 2019-07-23:
extfat4 should be self-healing with journal. Do you see other issues?

friendodevil, 2019-07-23:
Have noticed that sometimes HassOS gets stuck and requires a VM machine
restart,  not sure its related.

closed 2019-09-28 by stale[bot]

roybosch, 2020-03-20:
This seems to be closed, but I (and many others, so it seems) have this same problem:

https://community.home-assistant.io/t/ext4-fs-error-validating-inode-bitmap/163287/4

https://community.home-assistant.io/t/ext4-mb-generate-buddy-747/104860/6

Nothing is being self-healed, HA seems to get more and more corrupt until it doesn't boot at all.

blamaz, 2020-05-13:
I'm running vbox vm and also having the same problem to a point that HassOS just hang on reboot.  I have to do a vm api shutdown to resolve it. Cannot use reboot fonction for now.

roybosch, 2020-05-14:
> I'm running vbox vm and also having the same problem to a point that HassOS just hang on reboot. I have to do a vm api shutdown to resolve it. Cannot use reboot fonction for now. This issue is closed but problem remain.

What do you mean with a VM api shutdown? Is that a way to solve a corrupt VM?

blamaz, 2020-05-14:
> What do you mean with a VM api shutdown? Is that a way to solve a corrupt VM?

Sorry, I meant ACPI shutdown from the Virtualbox UI

iotinkognito, 2021-05-06:
Also still getting these issues.