home-assistant/operating-system #448: Stacktrace during boot - Raspberry Pi 4

Joeboyc2, 2019-07-31:
**Home Assistant release with the issue:**

Latest - 0.96.5
**Operating environment (HassOS/Generic):**

HassOS
**Supervisor logs:**

Unable to access

**Description of problem:**
I've recently migrated my install from a Raspi3 to a 4,.

I ran a fresh install on a new SD card and then restored a full snapshot, everything was working well and i went to move the pi to a new location, i shutdown using the shutdown option in the hassio console, and then started the pi up, it came back online but when i accessed the interface nothing loaded, i just had blue bar across the top.

I rebooted the pi a few more times (removing power and adding again), still no luck, i eventually plugged a hdmi cable in (i run the pi headless usually) and saw a stacktrace on the console, the device was not on the network and i have had to now re-flash the SD card and start again

[image]

[image]

pvizeli, 2019-07-31:
Look like a memory issue or so. Maybe fix with a firmware update of RPi4

Joeboyc2, 2019-08-15:
got it again, most annoying :(
[image]

closed 2019-08-24 by pvizeli