home-assistant/operating-system #454: DNS resolv.conf not propagating to home assistant container

ntompson, 2019-08-17:
Hi, I've been using home assistant on a hassos build on RPi3 for months without issue. I use a local DNS server (on a different physical machine), which resolves local hosts on my LAN. Just this morning, local domain name resolution started to fail inside the home assistant container, breaking a number of components that rely on local domain names. At the time, I was running HA 96.5 and hassos 2.10. Updating to HA 97.2 and hassos 2.12 made no difference.

I had been applying DNS settings using NetworkManager settings. Running `cat /etc/NetworkManager/system-connections/my-network` on hassos yields:
```
[connection]
id=hassos-network
uuid=<redacted>
type=ethernet
permissions=
timestamp=1558765264

[ethernet]
mac-address-blacklist=

[ipv4]
address1=10.0.0.133/24,10.0.0.138
dns=10.0.0.157;10.0.0.138;1.1.1.1;8.8.8.8;
dns-search=
ignore-auto-dns=true
method=manual

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=auto
```

`10.0.0.157` is my local DNS server and `10.0.0.138` is my router.

This successfully applies my DNS settings to `resolv.conf` within hassos:
```
# cat /etc/resolv.conf 
# Generated by NetworkManager
nameserver 10.0.0.157
nameserver 10.0.0.138
nameserver 1.1.1.1
# NOTE: the libc resolver may not support more than 3 nameservers.
# The nameservers listed below may not be recognized.
nameserver 8.8.8.8
```

However, if I do the same inside the `homeassistant` container:
```
# docker exec -it homeassistant bash
bash-5.0# cat /etc/resolv.conf 
search local.hass.io
nameserver 172.30.32.3
```
In the past, I could see that the additional `nameserver` entries had been applied as expected, but not any more. I see the same problem in the supervisor container.

After further investigation, I discovered the following in the hassos system log after a reboot:

```
starting version 3.2.8
18-06-22 11:12:16 INFO (MainThread) [__main__] Initialize Hass.io setup
18-06-22 11:12:16 INFO (SyncWorker_0) [hassio.docker.supervisor] Attach to Supervisor homeassistant/armhf-hassio-supervisor with version 179
18-06-22 11:12:17 INFO (MainThread) [__main__] Setup HassIO
18-06-22 11:12:17 INFO (SyncWorker_1) [hassio.docker.interface] Attach to homeassistant/armhf-hassio-dns with version 1
18-06-22 11:12:17 INFO (MainThread) [hassio.dns] Update resolv from Supervisor
18-06-22 11:12:17 ERROR (MainThread) [hassio.dns] Can't write local resolv: [Errno 13] Permission denied: '/etc/resolv.conf'
18-06-22 11:12:17 INFO (MainThread) [hassio.dns] Start CoreDNS plugin
18-06-22 11:12:17 INFO (MainThread) [hassio.misc.forwarder] Start DNS port forwarding to 172.30.32.3
18-06-22 11:12:17 INFO (SyncWorker_0) [hassio.docker.interface] Clean hassio_dns application
18-06-22 11:12:20 INFO (SyncWorker_0) [hassio.docker.dns] Start DNS homeassistant/armhf-hassio-dns with version 1
18-06-22 11:12:20 INFO (MainThread) [hassio.utils.gdbus] Introspect org.freedesktop.systemd1 on /org/freedesktop/systemd1
18-06-22 11:12:21 INFO (MainThread) [hassio.utils.gdbus] Connect to dbus: org.freedesktop.systemd1 - /org/freedesktop/systemd
...
```
Note the error complaining of a permission problem when writing to `resolv.conf`. 

My hassos build is absolutely standard and I've never monkeyed around with anything inside hassos. I can't think of anything that could have caused a permission error.

I currently have a temporary workaround, which is to run a `shell_script` to add the missing DNS address to `resolv.conf` triggered by an automation when HA starts up. But it's not a fabulous solution and it all seems a bit odd. So I'm stumped and suspecting an issue with hassos.

DrMarlin, 2019-08-17:
Likely related to this issue https://github.com/home-assistant/hassio/issues/1231

closed 2019-08-24 by pvizeli

ntompson, 2019-08-27:
I can confirm that my issue was resolved with recent updates to the hassio supervisor, which propagates NetworkManager DNS configuration through to the new CoreDNS container. All fixed.