home-assistant/operating-system #456: HassOS falling off network after 24 hours

nhorvath, 2019-08-20:
**Home Assistant release with the issue:**

0.97.2

**Operating environment (HassOS/Generic):**

hassio > os info
board: ova
version: "2.12"
version_cli: "12"
version_cli_latest: "12"
version_latest: "2.12"

supervisor 180

Running in x86_64 openbox vm

**Supervisor logs:**

not accessible

**Description of problem:**
approximately 24 hours after reboot hassos host loses connectivity. "ip a" on the host console shows no ip address assigned. This only started happening after the update that broke DNS.

nhorvath, 2019-08-20:
nmi-cli doesn't appear to exist but nmcli con show produces just this output (while still working):
# nmcli con show
NAME            UUID                                  TYPE      DEVICE 
HassOS default  f62bf7c2-e565-49ff-bbfc-a4cf791e6add  ethernet  enp0s3 

It dropped off after 24 hours again. nmcli con show was unchanged.

There's too much errors about connections after it fails in journalctl that anything that might have been about what happened when it started is out of the buffer.

I did use systemctl status NetworkManager to see where my lease file was and it appears the lease isn't even expired.

[image]

nhorvath, 2019-08-20:
This is the result of just `nmcli`

[image]

nhorvath, 2019-08-20:
some more nmcli status
[image]

running `systemclt restart NetworkManager` fixes the network
[image]

cogneato, 2019-08-20:
`nmcli` should also list dns configuration at the end of the last page. Were you able to see anything there?

nhorvath, 2019-08-20:
```
DNS configuration:
        servers: 192.168.75.1
        domains: horvath
        interface: enp0s3
```
i'll have to wait to tomorrow to see if it disappears.

nhorvath, 2019-08-20:
So I thought it was weird that there were 2 leases in my leases file. I checked it after restarting network manager and there are still 2 there, one looks right for my network compare with another box and the other has that october renew/expire time... not sure how it got there.
```
# cat /var/lib/NetworkManager/dhclient-f62bf7c2-e565-49ff-bbfc-a4cf791e6add-enp0s3.lease 
lease {
  interface "enp0s3";
  fixed-address 192.168.75.70;
  option subnet-mask 255.255.255.0;
  option dhcp-lease-time 86400;
  option routers 192.168.75.1;
  option dhcp-message-type 5;
  option dhcp-server-identifier 192.168.75.1;
  option domain-name-servers 192.168.75.1;
  option dhcp-renewal-time 43200;
  option dhcp-rebinding-time 75600;
  option broadcast-address 192.168.75.255;
  option host-name "hassio";
  option domain-name "horvath";
  renew 2 2019/10/22 19:55:22;
  rebind 3 2019/10/23 07:03:07;
  expire 3 2019/10/23 10:03:07;
}
lease {
  interface "enp0s3";
  fixed-address 192.168.75.70;
  option subnet-mask 255.255.255.0;
  option routers 192.168.75.1;
  option dhcp-lease-time 86400;
  option dhcp-message-type 5;
  option domain-name-servers 192.168.75.1;
  option dhcp-server-identifier 192.168.75.1;
  option dhcp-renewal-time 43200;
  option broadcast-address 192.168.75.255;
  option dhcp-rebinding-time 75600;
  option host-name "hassio";
  option domain-name "horvath";
  renew 3 2019/08/21 02:14:40;
  rebind 3 2019/08/21 13:09:03;
  expire 3 2019/08/21 16:09:03;
}
```
I'm going to try deleting my lease file and restarting network manager.

nhorvath, 2019-08-20:
I now have a normal looking lease file. I have no idea how those october dates could have gotten there but my guess is network manager saw it still had a valid lease so didn't bother renewing it. crossing my fingers I don't have a disconnect tomorrow around this time.

nhorvath, 2019-08-21:
I made it past the renew and rebind time without issue and see new leases in that file. I'm going to close this but I have no idea how NM would have put those future dates in the lease file.

closed 2019-08-21 by nhorvath