home-assistant/operating-system #458: Build failed for rpi4 and rpi4_64

iBobik, 2019-08-27:
I've tried to build current dev on clean Ubuntu VM following [instructions](https://github.com/home-assistant/hassos/blob/dev/Documentation/getting_started_development.md)

But it failed on a lot of errors:
[hassos-build.txt](https://github.com/home-assistant/hassos/files/3545930/hassos-build.txt)

iBobik, 2019-08-27:
Also failing with `make all`:

```
root@9f7393ea29d0:/build# make all
config opi_prime
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "opi_prime_defconfig"
make[1]: Entering directory '/build/buildroot'
#
# configuration written to /build/buildroot/.config
#
make[1]: Leaving directory '/build/buildroot'
build opi_prime
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
make[2]: Leaving directory '/build/buildroot'
>>> host-gcc-initial 7.4.0 Building
PATH="/build/buildroot/output/host/bin:/build/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin" PKG_CONFIG="/build/buildroot/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/build/buildroot/output/host/lib/pkgconfig:/build/buildroot/output/host/share/pkgconfig"  /usr/bin/make -j2 gcc_cv_libc_provides_ssp=yes all-gcc all-target-libgcc -C /build/buildroot/output/build/host-gcc-initial-7.4.0/build
make[2]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libiberty'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/intl'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/intl'
make[4]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libiberty/testsuite'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libiberty/testsuite'
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libiberty'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/libiberty'
make[4]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=/build/buildroot/output/host/bin/ccache /usr/bin/gcc" "CFLAGS=-O2 -I/build/buildroot/output/host/include" "CXXFLAGS=-O2 -I/build/buildroot/output/host/include" "CFLAGS_FOR_BUILD=-O2 -I/build/buildroot/output/host/include" "CFLAGS_FOR_TARGET=-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc -L/build/buildroot/output/host/lib -Wl,-rpath,/build/buildroot/output/host/lib" "LIBCFLAGS=-O2 -I/build/buildroot/output/host/include" "LIBCFLAGS_FOR_TARGET=-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " "MAKE=/usr/bin/make" "MAKEINFO=/build/buildroot/output/build/host-gcc-initial-7.4.0/missing makeinfo --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/build/buildroot/output/host" "infodir=/build/buildroot/output/host/share/info" "libdir=/build/buildroot/output/host/lib" "prefix=/build/buildroot/output/host" "tooldir=/build/buildroot/output/host/aarch64-buildroot-linux-gnu" "AR=/usr/bin/ar" "AS=/usr/bin/as" "CC=/build/buildroot/output/host/bin/ccache /usr/bin/gcc" "CXX=/build/buildroot/output/host/bin/ccache /usr/bin/g++" "LD=/usr/bin/ld" "LIBCFLAGS=-O2 -I/build/buildroot/output/host/include" "NM=/usr/bin/nm" "PICFLAG=" "RANLIB=/usr/bin/ranlib" "DESTDIR=" DO=all multi-do # /usr/bin/make
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/libiberty'
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/zlib'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libbacktrace'
/usr/bin/make  all-am
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libdecnumber'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libdecnumber'
make[4]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libbacktrace'
true  DO=all multi-do # /usr/bin/make
make[4]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libbacktrace'
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/libbacktrace'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/lto-plugin'
make[3]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/fixincludes'
/usr/bin/make  all-am
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/build-x86_64-pc-linux-gnu/fixincludes'
make[4]: Entering directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build/lto-plugin'
[... cut ...]
Makefile:4208: recipe for target 'all-gcc' failed
make[2]: *** [all-gcc] Error 2
make[2]: Leaving directory '/build/buildroot/output/build/host-gcc-initial-7.4.0/build'
package/pkg-generic.mk:238: recipe for target '/build/buildroot/output/build/host-gcc-initial-7.4.0/.stamp_built' failed
make[1]: *** [/build/buildroot/output/build/host-gcc-initial-7.4.0/.stamp_built] Error 2
make[1]: Leaving directory '/build/buildroot'
Makefile:24: recipe for target 'opi_prime' failed
make: *** [opi_prime] Error 2

```

pvizeli, 2019-08-27:
```
g++: internal compiler error: Killed (program cc1plus)
```

You need a lot more resource. 16GB memory and 4 cores with 2.8GHz are the min hardware requirement for the build process.

iBobik, 2019-08-27:
Thank you.

I've used 16 GB VM but it failed much sooner:

```
root@266165aaf5bb:/build# make rpi4
config rpi4
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "rpi4_defconfig"
make[1]: Entering directory '/build/buildroot'
#
# configuration written to /build/buildroot/.config
#
make[1]: Leaving directory '/build/buildroot'
build rpi4
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external
make[1]: Entering directory '/build/buildroot'
/usr/bin/make -j1 O=/build/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
make[2]: Leaving directory '/build/buildroot'
>>> glibc glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a Configuring
mkdir -p /build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/build
# Do the configuration
(cd /build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/build; PATH="/build/buildroot/output/host/bin:/build/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin" AR="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-ar" AS="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-as" LD="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-ld" NM="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-nm" CC="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc" GCC="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc" CPP="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-cpp" CXX="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-g++" FC="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gfortran" F77="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gfortran" RANLIB="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-ranlib" READELF="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-readelf" STRIP="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-strip" OBJCOPY="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-objcopy" OBJDUMP="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/build/buildroot/output/host/bin/ccache /usr/bin/gcc" GCC_FOR_BUILD="/build/buildroot/output/host/bin/ccache /usr/bin/gcc" CXX_FOR_BUILD="/build/buildroot/output/host/bin/ccache /usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/build/buildroot/output/host/include" CFLAGS_FOR_BUILD="-O2 -I/build/buildroot/output/host/include" CXXFLAGS_FOR_BUILD="-O2 -I/build/buildroot/output/host/include" LDFLAGS_FOR_BUILD="-L/build/buildroot/output/host/lib -Wl,-rpath,/build/buildroot/output/host/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-as" DEFAULT_LINKER="/build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-ld" CPPFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " LDFLAGS="" FCFLAGS=" -Os " FFLAGS=" -Os " PKG_CONFIG="/build/buildroot/output/host/bin/pkg-config" STAGING_DIR="/build/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot" INTLTOOL_PERL=/usr/bin/perl CFLAGS="-O2 " CPPFLAGS="" CXXFLAGS="-O2 " ac_cv_path_BASH_SHELL=/bin/bash libc_cv_forced_unwind=yes libc_cv_ssp=no ac_cv_prog_MAKE="/usr/bin/make -j7" /bin/bash /build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/configure --target=arm-buildroot-linux-gnueabihf --host=arm-buildroot-linux-gnueabihf --build=x86_64-pc-linux-gnu --prefix=/usr --enable-shared  --with-pkgversion="Buildroot" --without-cvs --disable-profile --without-gd --enable-obsolete-rpc --enable-kernel=4.19 --with-headers=/build/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include)
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-buildroot-linux-gnueabihf
checking for arm-buildroot-linux-gnueabihf-gcc... /build/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc
checking for suffix of object files... configure: error: in `/build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/build':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
package/pkg-generic.mk:228: recipe for target '/build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/.stamp_configured' failed
make[1]: *** [/build/buildroot/output/build/glibc-glibc-2.28-94-g4aeff335ca19286ee2382d8eba794ae5fd49281a/.stamp_configured] Error 1
make[1]: Leaving directory '/build/buildroot'
Makefile:24: recipe for target 'rpi4' failed
make: *** [rpi4] Error 2
```

pvizeli, 2019-08-28:
I don't have any issues on dev. I recommend reading the buildroot handbook first: https://buildroot.org/downloads/manual/manual.html

closed 2019-08-29 by pvizeli